pikvm/ustreamer #218: Low FPS when accesing server from outside LAN

Andrei-Iosifescu123, 2023-05-28:
I am running uStreamer on a Raspberry Pi 2B+. It works great when I am accesing it from LAN, but whenever I connect to it using VPN/port forwarding, I always get low (6-7) FPS compared to 25 locally. This is my command: `./ustreamer -r 1280x720 -m MJPEG -f 25 -s 0.0.0.0 -c M2M-IMAGE`.
Any help would be appreciated.

mdevaev, 2023-05-28:
Hello. The reason is the low bandwidth of the channel. MJPEG is a heavy format.

Andrei-Iosifescu123, 2023-05-29:
> Hello. The reason is the low bandwidth of the channel. MJPEG is a heavy format.

I have 100 Mbps connection on both sides, so this could not be the problem.

mdevaev, 2023-05-29:
You should check how big the delays are and whether the performance when transferring data from one end to the other is really fast. There really is nothing here that can be configured from the software side.

Andrei-Iosifescu123, 2023-06-03:
> You should check how big the delays are and whether the performance when transferring data from one end to the other is really fast. There really is nothing here that can be configured from the software side.

The average ping is about 35ms, and the speed is around 80Mbps. I think it aslo has something to do with `queued_fps`, since it is 24, and actual fps is 1. So while all frames are being encoded (`"captured_fps": 25`), not all reach the end device. 
[image]

mdevaev, 2023-06-03:
Which browser do you use?

Andrei-Iosifescu123, 2023-06-05:
> Which browser do you use?

For viewing the stream, I use an app called [IP Camera](https://play.google.com/store/apps/details?id=com.shenyaocn.android.WebCam)

mdevaev, 2023-06-05:
Please try the desktop browser like Chrome.

Andrei-Iosifescu123, 2023-06-28:
> Hello. The reason is the low bandwidth of the channel. MJPEG is a heavy format.

Is there any way to stream H264 instead of MJPG? It should have a much lower bandwidth.

mdevaev, 2023-06-28:
Yes, but not over HTTP.

Andrei-Iosifescu123, 2023-06-29:
> Yes, but not over HTTP.

Over RTSP? How can I do that? I didn't see any options regarding rtsp streaming.

mdevaev, 2023-06-29:
https://github.com/pikvm/ustreamer/blob/master/docs/h264.md

Andrei-Iosifescu123, 2023-07-04:
> https://github.com/pikvm/ustreamer/blob/master/docs/h264.md

I tried this, but when I ran `./ustreamer -r 1280x720 -f 25 --h264-sink demo::ustreamer::h264 --h264-sink-rm -s 0.0.0.0` I got an error:
```
-- ERROR [61288.860    stream] -- H264: Can't set option V4L2_CID_MPEG_VIDEO_H264_MIN_QP: Invalid argument
-- ERROR [61288.860    stream] -- H264: Encoder destroyed due an error (prepare)
```

mdevaev, 2023-07-05:
Is it raspberry pi?

Andrei-Iosifescu123, 2023-07-05:
> Is it raspberry pi?

Yes, Raspberry Pi 2B+ running Ubuntu 20.04

mdevaev, 2023-07-05:
Probably Ubuntu doesn't have all needed kernel patches to support it. Try the latest Raspberry OS.

Andrei-Iosifescu123, 2023-07-07:
> Probably Ubuntu doesn't have all needed kernel patches to support it. Try the latest Raspberry OS.

But since it's a USB camera it should work on any kind of computer running Ubuntu - not only Raspberry Pis.

mdevaev, 2023-07-07:
No because this is not related with camera, but with hardware h264 encoder.

Also you need to use yuyv or uyvy for the input format, not mjpeg/jpeg, otherwise the image from camera will be decoded on cpu befure h264 encoding.

Andrei-Iosifescu123, 2023-07-08:
> No because this is not related with camera, but with hardware h264 encoder.
> 
> Also you need to use yuyv or uyvy for the input format, not mjpeg/jpeg, otherwise the image from camera will be decoded on cpu befure h264 encoding.

The problem is that my camera can only do 10FPS using yuyv image format and 25 using MJPG. Is it possible to encode the MJPG from the camera to a H264 stream using hardware acceleration?

mdevaev, 2023-07-08:
Yes but it will slow.

Andrei-Iosifescu123, 2023-07-08:
> Yes but it will slow.

So there's no way to get 25 FPS out of this camera using uStreamer and Janus?

mdevaev, 2023-07-08:
No, only for specific capture devices or cameras. For example, it should work for Camera Module 3 (because of CSI bus), but I didn't test it. Also it will work with HDMI capture. But USB not for USB camera I think.

Andrei-Iosifescu123, 2023-07-09:
> No, only for specific capture devices or cameras. For example, it should work for Camera Module 3 (because of CSI bus), but I didn't test it. Also it will work with HDMI capture. But USB not for USB camera I think.

I think i'll just stick with ffmpeg for now. Thanks for your help.

closed 2023-07-09 by Andrei-Iosifescu123

mdevaev, 2023-07-09:
You're welcome. I'm sorry it didn't suit you.