pikvm/ustreamer #219: Mainloop select() error: No error information When running on 1920x1080p resolution

11philip22, 2023-06-13:
When I run `./ustreamer --device=/dev/video0 --host=0.0.0.0 --resolution 1920x1080 --quality 100` I get the error `Mainloop select() error: No error information`
This is the output of `v4l2-ctl --list-formats-ext`
```
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1280x960
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
```
I cant try MJPG since the device I got has no dedicated gpu and from what I understood thats needed.
But even if I specify 5 fps I still get the error.

mdevaev, 2023-06-13:
Hello. Please attach the full log.

11philip22, 2023-06-14:
@mdevaev Here is the debug log
```
~/ustreamer # ./ustreamer --device=/dev/video0 --host=0.0.0.0 --port=80 --quality 100 --desired-fps 5 --resolution 1920x1080 --log-level 3
-- INFO  [92081.528      main] -- Starting PiKVM uStreamer 5.40 ...
-- INFO  [92081.529      main] -- Using internal blank placeholder
-- DEBUG [92081.529      main] -- Installing SIGINT handler ...
-- DEBUG [92081.529      main] -- Installing SIGTERM handler ...
-- DEBUG [92081.529      main] -- Ignoring SIGPIPE ...
-- DEBUG [92081.529      main] -- Binding HTTP to [0.0.0.0]:80 ...
-- INFO  [92081.529      main] -- Listening HTTP on [0.0.0.0]:80
-- INFO  [92081.530    stream] -- Using V4L2 device: /dev/video0
-- INFO  [92081.530      http] -- Starting HTTP eventloop ...
-- INFO  [92081.530    stream] -- Using desired FPS: 5
-- DEBUG [92081.530    stream] -- _stream_init_loop: stream->run->stop=0
================================================================================
-- INFO  [92081.595    stream] -- Device fd=8 opened
-- DEBUG [92081.596    stream] -- Querying device capabilities ...
-- INFO  [92081.596    stream] -- Using input channel: 0
-- DEBUG [92081.596    stream] -- Using TV standard: DEFAULT
-- DEBUG [92081.596    stream] -- Probing device format=YUYV, stride=3840, resolution=1920x1080 ...
-- INFO  [92081.600    stream] -- Using resolution: 1920x1080
-- INFO  [92081.600    stream] -- Using format: YUYV
-- DEBUG [92081.600    stream] -- Querying HW FPS ...
-- INFO  [92081.602    stream] -- Using HW FPS: 5
-- INFO  [92081.602    stream] -- Using IO method: MMAP
-- DEBUG [92081.602    stream] -- Requesting 3 device buffers for MMAP ...
-- INFO  [92081.613    stream] -- Requested 3 device buffers, got 3
-- DEBUG [92081.613    stream] -- Allocating device buffers ...
-- DEBUG [92081.613    stream] -- Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=0 ...
-- DEBUG [92081.613    stream] -- Mapping device buffer=0 ...
-- DEBUG [92081.613    stream] -- Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=1 ...
-- DEBUG [92081.613    stream] -- Mapping device buffer=1 ...
-- DEBUG [92081.614    stream] -- Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=2 ...
-- DEBUG [92081.614    stream] -- Mapping device buffer=2 ...
-- DEBUG [92081.615    stream] -- Calling us_xioctl(VIDIOC_QBUF) for buffer=0 ...
-- DEBUG [92081.615    stream] -- Calling us_xioctl(VIDIOC_QBUF) for buffer=1 ...
-- DEBUG [92081.615    stream] -- Calling us_xioctl(VIDIOC_QBUF) for buffer=2 ...
-- DEBUG [92081.615    stream] -- Device fd=8 initialized
-- DEBUG [92081.615    stream] -- Starting device capturing ...
-- DEBUG [92081.629      http] -- HTTP: Updating exposed frame (online=0) ...
-- VERB  [92081.629      http] -- HTTP: Exposed frame: online=0, exp_time=0.000000
-- INFO  [92082.256    stream] -- Capturing started
-- INFO  [92082.256    stream] -- Using JPEG quality: 100%
-- INFO  [92082.256    stream] -- Creating pool JPEG with 2 workers ...
-- DEBUG [92082.257      jw-0] -- Hello! I am a worker jw-0 ^_^
-- DEBUG [92082.257      jw-0] -- Worker jw-0 waiting for a new job ...
-- DEBUG [92082.257      jw-1] -- Hello! I am a worker jw-1 ^_^
-- INFO  [92082.257    stream] -- Capturing ...
--------------------------------------------------------------------------------
-- DEBUG [92082.259    stream] -- Waiting for worker ...
-- DEBUG [92082.259    stream] -- Calling select() on video device ...
-- DEBUG [92082.259      jw-1] -- Worker jw-1 waiting for a new job ...
-- DEBUG [92082.730      http] -- HTTP: Repeating exposed ...
-- DEBUG [92083.260    stream] -- Device select() --> 0
-- ERROR [92083.260    stream] -- Mainloop select() error: No error information
-- INFO  [92083.260    stream] -- Destroying workers pool JPEG ...
-- DEBUG [92083.260      jw-0] -- Bye-bye (worker jw-0)
-- DEBUG [92083.261      jw-1] -- Bye-bye (worker jw-1)
-- DEBUG [92083.261    stream] -- Stopping device capturing ...
-- INFO  [92083.498    stream] -- Capturing stopped
-- DEBUG [92083.498    stream] -- Releasing device buffers ...
-- DEBUG [92083.499    stream] -- Closing device ...
-- INFO  [92083.502    stream] -- Device fd=8 closed
-- DEBUG [92083.502    stream] -- _stream_init_loop: stream->run->stop=0
```

mdevaev, 2023-06-14:
Try option `-m mjpeg`

11philip22, 2023-06-15:
That works :) Thanks. I initially was confused Because it still said:  
 `ERROR [170095.288    stream] -- Device doesn't support setting of HW encoding quality parameters`   
which is kinda weird since I only specify resolution:   
`./ustreamer --device=/dev/video0 --host=0.0.0.0 --port=80 --resolution 1920x1080 -m mjpeg`

mdevaev, 2023-06-15:
:ok_hand: