pikvm/ustreamer #223: ustreamer error at higher resolution (1600x1200)

conix67, 2023-07-19:
Using no-name USB camera /w support for 1600x1200 and getting following error. Can't tell what's wrong with the error message alone.. ustreamer v5.24 on RPI4 (currently in use for PIKVM)

================================================================================
-- INFO  [2277.756    stream] -- Device fd=8 opened
-- INFO  [2277.757    stream] -- Using input channel: 0
-- INFO  [2277.809    stream] -- Using resolution: 1600x1200
-- INFO  [2277.809    stream] -- Using format: YUYV
-- INFO  [2277.813    stream] -- Using HW FPS: 0 -> 5 (coerced)
-- INFO  [2277.813    stream] -- Using IO method: MMAP
-- INFO  [2277.841    stream] -- Requested 5 device buffers, got 5
-- INFO  [2277.858    stream] -- Capturing started
-- INFO  [2277.858    stream] -- Using JPEG quality: 80%
-- INFO  [2277.858    stream] -- Creating pool JPEG with 4 workers ...
-- INFO  [2277.858    stream] -- Capturing ...
**-- ERROR [2278.859    stream] -- Mainloop select() error: Success**
-- INFO  [2278.860    stream] -- Destroying workers pool JPEG ...
-- INFO  [2278.863    stream] -- Capturing stopped
-- INFO  [2278.867    stream] -- Device fd=8 closed
================================================================================

The device works fine at next lower resolution - 800x600.

mdevaev, 2023-07-19:
Try `-m jpeg` or `-m mjpeg`.

conix67, 2023-07-19:
Thanks for the quick response! I'm getting below with -m jpeg or -m mjpeg
-- ERROR [3439.299    stream] -- Could not obtain the requested format=MJPEG; driver gave us YUYV

mdevaev, 2023-07-20:
Show me `v4l2-ctl --list-formats-ext`

conix67, 2023-07-24:
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1600x1200
                        Interval: Discrete 0.200s (5.000 fps)

mdevaev, 2023-07-24:
1600x1200 is a maximum supported resolution by your cam.

conix67, 2023-07-25:
Well that makes sense. I was confused by the error message "Success".

mdevaev, 2023-07-25:
This is due to the fact that errno was not set properly in syscall. Most likely the fault of the driver that does not handle this situation.