pikvm/ustreamer #224: Feature Request: Enable HTTPS/SSL Certificates on Stream Site

tallman5, 2023-07-24:
Love the low CPU usage! Thank you!

I have the port opened to the video stream available on the internet. Stream shows up properly. However, browsers these days are blocking mixed content (HTTP/HTTPS) by default. If I have an HTTPS page, the video source cannot be HTTP. In the browser settings, I can whitelist the site to allow mixed content.

I can help with getting a valid cert onto a Raspberry Pi. However, I have no idea how to incorporate it into your web server.

Thanks again!

mdevaev, 2023-07-24:
Hello!

I think the best solution would be to configure Nginx as a reverse proxy for uStreamer. SSL is a fairly extensive topic and it is better to entrust it to a component that copes with it perfectly.

tallman5, 2023-07-28:
> Hello!
> 
> I think the best solution would be to configure Nginx as a reverse proxy for uStreamer. SSL is a fairly extensive topic and it is better to entrust it to a component that copes with it perfectly.

10-4. I added a PR for setting up the proxy.

closed 2023-07-28 by tallman5