pikvm/ustreamer #225: upgrade to libgpio-2.0

tao-j, 2023-07-27:
arch still has the 1.6 https://aur.archlinux.org/packages/libgpiod-git

fedora is moving to 2.0 https://src.fedoraproject.org/rpms/libgpiod

2.0 breaks some api and according to my initial research

mdevaev, 2023-08-07:
I'll do it when Arch gets libgpiod2.

mdevaev, 2024-01-07:
Added in master branch.

closed 2024-01-07 by mdevaev

jerry-yuan, 2024-05-25:
@mdevaev Have you fixed this? I was failed for the reason that arch provided libgpiod with v2 when building ustreamer with `makepkg -si` at `pkg/arch`:
[image]
And I have confirmed that ustreamer still using libgpiod v1 apis and my libgpiod has been upgraded into v2:
[image]

jerry-yuan, 2024-05-25:
> @mdevaev Have you fixed this? I was failed for the reason that arch provided libgpiod with v2 when building ustreamer with `makepkg -si` at `pkg/arch`: [...]

I have solved this, I checked the commits and I found you added a macro `HAVE_GPIOD2` in this [commit](https://github.com/pikvm/ustreamer/commit/d9b5f2b03d079396548f71af3c2036b8433c6961) by execute `pkg-config` to check if the libgpiod v2 is exist. But in my case the `pkg-config` is not installed, which will make this always fails and down-grade to use libgpiod v1 apis(I was build this in a clear docker container). Maybe you should add a check of pkg-config's existance?

mdevaev, 2024-05-26:
It makes sense. Added.