pikvm/ustreamer #227: question: privacy mode

zerog2k, 2023-07-31:
Hi, first off thanks for ustreamer!
I'm using it with octoprint to view 3d printer activity.

I'm curious if there is a way (api call, signal, etc) to disable the camera (turn it off, and optionally just serve blank/black frames, or something else) when not in use. Like most USB webcams, mine has an led status light to indicate that it's currently active. Because ustreamer is always running, it seems like the camera is always active, even when we are not printing anything.

I'd like to be able to write a plugin or extend octoprint functionality to turn on/off the camera in the octoprint UI.

Does anything like this exist already?

mdevaev, 2023-08-01:
Hello. I haven't heard of anything like this. It is assumed that ustreamer does not contain any real-time configuration and management tools, and if you need to stop a stream, you stop the service. This is not very convenient in the general case, but it was quite enough for PiKVM. But I think you can use socket activation & timeout to stop: https://github.com/pikvm/ustreamer/issues/16

mdevaev, 2023-08-04:
Answered, I guess.

closed 2023-08-04 by mdevaev