pikvm/ustreamer #231: Video capture is not supported by device

ProTechEx, 2023-09-20:
How I can solve this issue:

```
[2023-09-20 11:10:17 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2023-09-20 11:10:17 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [222.904    stream] -- Device fd=10 opened
[2023-09-20 11:10:17 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- ERROR [222.904    stream] -- Video capture is not supported by device
[2023-09-20 11:10:17 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [222.904    stream] -- Device fd=10 closed
[2023-09-20 11:10:17 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [222.904    stream] -- Sleeping 1 seconds before new stream init ...
[2023-09-20 11:10:18 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2023-09-20 11:10:18 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [223.905    stream] -- Device fd=10 opened
[2023-09-20 11:10:18 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- ERROR [223.905    stream] -- Video capture is not supported by device
[2023-09-20 11:10:18 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [223.905    stream] -- Device fd=10 closed
[2023-09-20 11:10:18 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [223.905    stream] -- Sleeping 1 seconds before new stream init ...
[2023-09-20 11:10:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2023-09-20 11:10:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [224.906    stream] -- Device fd=10 opened
[2023-09-20 11:10:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- ERROR [224.906    stream] -- Video capture is not supported by device
[2023-09-20 11:10:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [224.906    stream] -- Device fd=10 closed
[2023-09-20 11:10:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [224.906    stream] -- Sleeping 1 seconds before new stream init ...
[2023-09-20 11:10:20 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2023-09-20 11:10:20 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [225.906    stream] -- Device fd=10 opened
[2023-09-20 11:10:20 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- ERROR [225.907    stream] -- Video capture is not supported by device
[2023-09-20 11:10:20 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [225.907    stream] -- Device fd=10 closed
[2023-09-20 11:10:20 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [225.907    stream] -- Sleeping 1 seconds before new stream init ...
[2023-09-20 11:10:21 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2023-09-20 11:10:21 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [226.907    stream] -- Device fd=10 opened
[2023-09-20 11:10:21 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- ERROR [226.907    stream] -- Video capture is not supported by device
[2023-09-20 11:10:21 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [226.907    stream] -- Device fd=10 closed
[2023-09-20 11:10:21 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [226.908    stream] -- Sleeping 1 seconds before new stream init ...
[2023-09-20 11:10:22 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2023-09-20 11:10:22 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [227.908    stream] -- Device fd=10 opened
[2023-09-20 11:10:22 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- ERROR [227.908    stream] -- Video capture is not supported by device
[2023-09-20 11:10:22 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [227.908    stream] -- Device fd=10 closed
[2023-09-20 11:10:22 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [227.908    stream] -- Sleeping 1 seconds before new stream init ...
[2023-09-20 11:10:23 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2023-09-20 11:10:23 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [228.909    stream] -- Device fd=10 opened
[2023-09-20 11:10:23 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- ERROR [228.909    stream] -- Video capture is not supported by device
[2023-09-20 11:10:23 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [228.909    stream] -- Device fd=10 closed
[2023-09-20 11:10:23 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [228.909    stream] -- Sleeping 1 seconds before new stream init ...
[2023-09-20 11:10:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2023-09-20 11:10:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [229.910    stream] -- Device fd=10 opened
[2023-09-20 11:10:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- ERROR [229.910    stream] -- Video capture is not supported by device
[2023-09-20 11:10:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [229.910    stream] -- Device fd=10 closed
[2023-09-20 11:10:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [229.910    stream] -- Sleeping 1 seconds before new stream init ...
[2023-09-20 11:10:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Stopping streamer after delay ...
[2023-09-20 11:10:25 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Process killed: retcode=0
```

```
[kvmd-webterm@pikvm ~]$ dmesg | grep video
[    9.401559] videodev: Linux video capture interface: v2.00
[    9.572750] rpivid feb10000.codec: Device registered as /dev/video19
[    9.594652] usbcore: registered new interface driver uvcvideo
[    9.601517] (efault): Unregistering node (null)[0] device node /dev/video0
[    9.601526] (efault): Unregistering node (null)[0] device node /dev/video0
[    9.601530] (efault): Unregistering node (null)[0] device node /dev/video0
[    9.601533] (efault): Unregistering node (null)[0] device node /dev/video0
```

```
[kvmd-webterm@pikvm ~]$ v4l2-ctl --all
Driver Info:
        Driver name      : uvcvideo
        Card type        : USB3. 0 capture: USB3. 0 captur
        Bus info         : usb-0000:01:00.0-1.4
        Driver version   : 5.15.68
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height      : 1920/1080
        Pixel Format      : 'MJPG' (Motion-JPEG)
        Field             : None
        Bytes per Line    : 0
        Size Image        : 4147200
        Colorspace        : sRGB
        Transfer Function : Rec. 709
        YCbCr/HSV Encoding: ITU-R 601
        Quantization      : Default (maps to Full Range)
        Flags             : 
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 1920, Height 1080
        Default     : Left 0, Top 0, Width 1920, Height 1080
        Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Read buffers     : 0
                     brightness 0x00980900 (int)    : min=-128 max=127 step=1 default=-11 value=-11
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=148 value=148
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=180 value=180
                            hue 0x00980903 (int)    : min=-128 max=127 step=1 default=0 value=0
```

ProTechEx, 2023-09-20:
I get "No Signal"

[image]

But it had worked 100% in the past.

mdevaev, 2023-09-20:
Hello. This is most likely a driver problem. Try any other video capture software to confirm or deny this.

ProTechEx, 2023-09-20:
Thanks for reply, but I don't understand what exactly you mean with "Try any video capture software".
Can you explain that part or is there any how to regarding that ?

mdevaev, 2023-09-27:
> After flashing the image it had worked. I believe the os has updated somehow the driver.

Did you update OS after installation manually?

mdevaev, 2024-02-03:
No answer here.

closed 2024-02-03 by mdevaev