pikvm/ustreamer #232: Stream audio from non-hdmi interface

samdelong, 2023-09-20:
I have a podtrak mic/recorder plugged into my pikvm and host machine. On the PIKVM, the podtrak acts as an audio input like the currently supported HDMI input. Can I set my janus/ustreamer audio device to the podtrak input instead of hdmi? So I can hear host machine audio.
I can't use HDMI audio because the host can only use one audio device at a time, and it needs to be the Podtrak.
Ive tried changing the audio device in /etc/kvmd/janus/janus.plugin.ustreamer.jcfg, but no luck.

samdelong, 2023-09-29:
I checked the janus log, and it doesn't appear to be erroring. However video and audio stops working entirely
My janus config:
```
video: {
    sink = "kvmd::ustreamer::h264"
}
audio: {
    device = "hw:1" # switched from hw:0,0
    tc358743 = "/dev/kvmd-video"
}
```
My janus log:
```
Sep 10 17:02:49 pikvm systemd[1]: Started PiKVM - Janus WebRTC Gateway.
Sep 10 17:02:49 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- Starting Janus Runner ...
Sep 10 17:02:49 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- Probbing the network first time ...
Sep 29 18:47:53 pikvm kvmd-janus[598]: kvmd.apps.janus.stun             ERROR --- First probe: Can't perform STUN request after 5 retries; last error: Send error: OSError: [Errno 101] Network is unreachable
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- Got new _Netcfg(nat_type='Blocked', src_ip='fe80::dea6:32ff:fe4c:ecbb%wlan0', ext_ip='', stun_host='stun.l.google.com', stun_port=19302)
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- Starting Janus ...
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- Started Janus pid=761: /usr/bin/janus --disable-colors --plugins-folder=/usr/lib/ustreamer/janus --configs-folder=/etc/kvmd/janus --interface=fe80::dea6:32ff:f>
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Janus commit: not-a-git-repo
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Compiled on:  Thu Jun 29 22:53:31 UTC 2023
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Logger plugins folder: /usr/lib/janus/loggers
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => ---------------------------------------------------
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Starting Meetecho Janus (WebRTC Server) v0.14.0
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => ---------------------------------------------------
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Checking command line arguments...
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Debug/log level is 4
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Debug/log timestamps are disabled
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Debug/log colors are disabled
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => [WARN] Error setting local IP address to fe80::dea6:32ff:fe4c:ecbb%wlan0, falling back to detecting IP address...
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Using 10.7.1.113 as local IP...
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Token based authentication disabled
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Initializing recorder code
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => RTP port range: 20000 -- 40000
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Initializing ICE stuff (Full mode, ICE-TCP candidates disabled, half-trickle, IPv6 support enabled)
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => ICE port range: 20000-40000
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => [WARN] mDNS resolution disabled, .local candidates will be ignored
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => [WARN] Janus is deployed on a private address (10.7.1.113) but you didn't specify any STUN server! Expect trouble if this is supposed to work over the inter>
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Crypto: OpenSSL >= 1.1.0
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => No cert/key specified, autogenerating some...
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Fingerprint of our certificate: 33:32:39:5B:54:DA:7E:88:54:73:E3:FA:EB:DB:C8:4A:D6:61:A6:4D:C4:73:E7:7D:E4:06:0F:CC:97:B2:65:14
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => [WARN] Data Channels support not compiled
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Sessions watchdog started
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Event handlers support disabled
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Plugins folder: /usr/lib/ustreamer/janus
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Joining Janus requests handler thread
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Loading plugin 'libjanus_ustreamer.so'...
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => == ustreamer/main      -- Initializing PiKVM uStreamer plugin 5.42 ...
Sep 29 18:47:58 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => == ustreamer/config    -- Reading config file '/etc/kvmd/janus/janus.plugin.ustreamer.jcfg' ...
[... cut ...]
Sep 29 18:48:13 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Janus commit: not-a-git-repo
Sep 29 18:48:13 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Compiled on:  Thu Jun 29 22:53:31 UTC 2023
Sep 29 18:48:13 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => Logger plugins folder: /usr/lib/janus/loggers
Sep 29 18:48:13 pikvm kvmd-janus[598]: kvmd.apps.janus.runner            INFO --- => ---------------------------------------------------```

samdelong, 2023-09-29:
Is it possible that I need to change the HDMI audio device in `/boot/config.txt` ?

mdevaev, 2023-09-30:
No. Sorry, I know what's wrong. Our plugin for Janus does not implement the usual configuration of audio capture devices. I'll do it someday, but I can't promise it will be fast.

samdelong, 2023-10-01:
Ok I see, thanks. Can you point me to the janus plugin part of the codebase? I'll give a shot at implementing it.

closed 2023-10-01 by samdelong

mdevaev, 2023-10-02:
It's here: https://github.com/pikvm/ustreamer/tree/master/janus/src