pikvm/ustreamer #234: Can't set input channel BTT-CB1 (BTT-CB1 2.3.3 Bullseye with Linux 5.16.17-sun50iw9).

hulecomte, 2023-10-07:
Hey, 

for some reasons I can't set an input channel for my usb camera on my BTT-CB1 card (pi clone).

I already mentioned this on the PiKVM discord and figured out I would make a proper issue for it! If needed I would be super down for helping out on this if someone could point me to the right direction!

I did try this patch: #233 but without any success so far.

`biqu@BTT-CB1:~/src/ustreamer$ lsb_release 
No LSB modules are available.
biqu@BTT-CB1:~/src/ustreamer$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
biqu@BTT-CB1:~/src/ustreamer$ v4l2-ctl --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : USB HD Camera: USB HD Camera
	Bus info         : usb-5200000.usb-1
	Driver version   : 5.16.17
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : USB HD Camera: USB HD Camera
	Serial           : 200910120001
	Bus info         : usb-5200000.usb-1
	Media version    : 5.16.17
	Hardware revision: 0x00001601 (5633)
	Driver version   : 5.16.17
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : USB HD Camera: USB HD Camera
	Function         : V4L2 I/O
	Flags         : default
	Pad 0x01000007   : 0: Sink
	  Link 0x02000010: from remote pad 0x100000a of entity 'Extension 4': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'MJPG' (Motion-JPEG)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 614400
	Colorspace        : sRGB
	Transfer Function : Rec. 709
	YCbCr/HSV Encoding: ITU-R 601
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 480
	Default     : Left 0, Top 0, Width 640, Height 480
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 640, Height 480, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 480, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0

User Controls

                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=100 step=1 default=38 value=38
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=64 value=64
                            hue 0x00980903 (int)    : min=-180 max=180 step=1 default=0 value=0
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=500 step=1 default=400 value=400
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
				0: Disabled
				1: 50 Hz
				2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=10 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=100 step=1 default=93 value=93
         backlight_compensation 0x0098091c (int)    : min=0 max=121 step=1 default=1 value=1

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
				1: Manual Mode
				3: Aperture Priority Mode
         exposure_time_absolute 0x009a0902 (int)    : min=50 max=10000 step=1 default=166 value=166 flags=inactive
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=1
                   pan_absolute 0x009a0908 (int)    : min=-57600 max=57600 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-43200 max=43200 step=3600 default=0 value=0
                  zoom_absolute 0x009a090d (int)    : min=0 max=3 step=1 default=0 value=0
biqu@BTT-CB1:~/src/ustreamer$ ./ustreamer --device=/dev/video1 --host=192.168.0.75 --port=8080
-- INFO  [810.100      main] -- Starting PiKVM uStreamer 5.43 ...
-- INFO  [810.100      main] -- Using internal blank placeholder
-- INFO  [810.101      main] -- Listening HTTP on [192.168.0.75]:8080
-- INFO  [810.102    stream] -- Using V4L2 device: /dev/video1
-- INFO  [810.102    stream] -- Using desired FPS: 0
================================================================================
-- INFO  [810.103      http] -- Starting HTTP eventloop ...
-- INFO  [810.103    stream] -- Device fd=8 opened
-- INFO  [810.103    stream] -- Using input channel: 0
-- ERROR [810.103    stream] -- Can't set input channel
-- INFO  [810.103    stream] -- Device fd=8 closed
-- INFO  [810.103    stream] -- Sleeping 1 seconds before new stream init ...
================================================================================
`

What do you think?

argakon, 2023-10-07:
Hello, the PR #233 doesn't help you because your device is not a multi planar. 
This is a input channel set problem. Could you trye a diffirent input channel? e.g. `-i 1`

argakon, 2023-10-07:
If it still doesn't work for you, try to comment out lines 488-493 and rebuild. https://github.com/pikvm/ustreamer/blob/master/src/ustreamer/device.c#L488

hulecomte, 2023-10-07:
> If it still doesn't work for you, try to comment out lines 488-493 and rebuild. https://github.com/pikvm/ustreamer/blob/master/src/ustreamer/device.c#L488

Hey @argakon thanks a bunch this suggestion worked for me!

```
-- ERROR [462.763    stream] -- Can't set device format: Invalid argument

```

Now I get this error. I can't really figure out what device format is about here, is it the "--device" option or something else?

Thanks in advance!

argakon, 2023-10-07:
> -- ERROR [462.763    stream] -- Can't set device format: Invalid argument

Add `--debug` to your cmd line, drop here a bit more info.

argakon, 2023-10-07:
I guess you need to set pixel format to mjpg in cmd line.

hulecomte, 2023-10-07:
```
```

mdevaev, 2023-10-10:
Will it work?
```
sudo ./ustreamer --device=/dev/video0 --host=192.168.0.75 --port=8080 --debug --format=MJPEG --resolution=640x480
```

hulecomte, 2023-10-11:
Hi, nope, still got the same error:
```
biqu@BTT-CB1:~/src/ustreamer$ sudo ./ustreamer --device=/dev/video0 --host=192.168.0.75 --port=8080 --debug --format=MJPEG --resolution=640x480
-- INFO  [138.079      main] -- Starting PiKVM uStreamer 5.43 ...
-- INFO  [138.079      main] -- Using internal blank placeholder
-- DEBUG [138.080      main] -- Installing SIGINT handler ...
-- DEBUG [138.080      main] -- Installing SIGTERM handler ...
-- DEBUG [138.080      main] -- Ignoring SIGPIPE ...
-- DEBUG [138.080      main] -- Binding HTTP to [192.168.0.75]:8080 ...
-- INFO  [138.081      main] -- Listening HTTP on [192.168.0.75]:8080
-- INFO  [138.082    stream] -- Using V4L2 device: /dev/video0
-- INFO  [138.082    stream] -- Using desired FPS: 0
-- DEBUG [138.082    stream] -- _stream_init_loop: stream->run->stop=0
-- INFO  [138.082      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [138.084    stream] -- Device fd=8 opened
-- DEBUG [138.084    stream] -- Querying device capabilities ...
-- DEBUG [138.084    stream] -- Using TV standard: DEFAULT
-- DEBUG [138.084    stream] -- Probing device format=MJPEG, stride=1280, resolution=640x480 ...
-- ERROR [138.084    stream] -- Can't set device format: Invalid argument
-- DEBUG [138.084    stream] -- Closing device ...
-- INFO  [138.085    stream] -- Device fd=8 closed
-- INFO  [138.085    stream] -- Sleeping 1 seconds before new stream init ...
-- DEBUG [138.095      http] -- HTTP: Updating exposed frame (online=0) ...
-- VERB  [138.095      http] -- HTTP: Exposed frame: online=0, exp_time=0.000000
================================================================================

```

mdevaev, 2023-10-11:
Probably something wrong with the driver. Anyway, I don't have board to test it.

hulecomte, 2023-10-21:
Hey, thanks for your answer. I'm motivated to make this work, would you have any lead to test what's relevant?

Also I managed to take a picture with the camera using ```
fswebcam -r 640x480 --jpeg 85 -D 1 web-cam-shot.jpg```

mdevaev, 2023-10-21:
I don't know what I can do here.

hulecomte, 2023-10-21:
I managed to make it work with mjpg-streamer! So the drivers are probably not the issue here right?

Still want to try ustreamer though!

argakon, 2023-10-21:
```
It captures a number
of frames from any V4L or V4L2 compatible device
```

did you checked which api your devices uses?

hulecomte, 2023-10-21:
It's compatible with V4L2:
```
biqu@BTT-CB1:~$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)

```

Now, idk why I get the device format error with ustreamer. I hade a similar error on `mjpg-streamer` when the device was occupied with `motion` deamon (I tried to setup this one prior to `mjpg-streamer`). But I see nothing when trying to see if some process is using the device.

argakon, 2023-10-21:
I mean, could you run `fswebcam` with debug flags and try to figure out which api is in use?

hulecomte, 2023-10-21:
I used the -v flag (didn't saw other debug flag than the verbosity mode):

```
biqu@BTT-CB1:~$ fswebcam -r 640x480 --jpeg 100 -D 1 -v web-cam-shot.jpg
--- Opening /dev/video0...
Trying source module v4l2...
/dev/video0 opened.
src_v4l2_get_capability,87: /dev/video0 information:
src_v4l2_get_capability,88: cap.driver: "uvcvideo"
src_v4l2_get_capability,89: cap.card: "USB HD Camera: USB HD Camera"
src_v4l2_get_capability,90: cap.bus_info: "usb-5200000.usb-1"
src_v4l2_get_capability,91: cap.capabilities=0x84A00001
src_v4l2_get_capability,92: - VIDEO_CAPTURE
src_v4l2_get_capability,103: - STREAMING
No input was specified, using the first.
src_v4l2_set_input,181: /dev/video0: Input 0 information:
src_v4l2_set_input,182: name = "Camera 1"
src_v4l2_set_input,183: type = 00000002
src_v4l2_set_input,185: - CAMERA
src_v4l2_set_input,186: audioset = 00000000
src_v4l2_set_input,187: tuner = 00000000
src_v4l2_set_input,188: status = 00000000
src_v4l2_set_pix_format,520: Device offers the following V4L2 pixel formats:
src_v4l2_set_pix_format,533: 0: [0x47504A4D] 'MJPG' (Motion-JPEG)
src_v4l2_set_pix_format,533: 1: [0x56595559] 'YUYV' (YUYV 4:2:2)
Using palette MJPEG
Delaying 1 seconds.
src_v4l2_set_mmap,672: mmap information:
src_v4l2_set_mmap,673: frames=4
src_v4l2_set_mmap,722: 0 length=614400
src_v4l2_set_mmap,722: 1 length=614400
src_v4l2_set_mmap,722: 2 length=614400
src_v4l2_set_mmap,722: 3 length=614400
--- Capturing frame...
Captured frame in 0.00 seconds.
--- Processing captured image...
Setting output format to JPEG, quality 100
Writing JPEG image to 'web-cam-shot.jpg'.
```
v4l2 compatible right?

mdevaev, 2023-10-21:
Try to comment this line and rebuild ustreamer: https://github.com/pikvm/ustreamer/blob/master/src/ustreamer/device.c#L628

mdevaev, 2023-10-25:
Sup?

hulecomte, 2023-10-26:
Hey! Thanks for checking in, I will be able to test this tomorrow!

mdevaev, 2023-11-15:
@hulecomte sup. Did you test it?

mdevaev, 2024-02-03:
Sup?

mdevaev, 2024-02-08:
No answer, closing.

closed 2024-02-08 by mdevaev