pikvm/ustreamer #238: Unable to start capturing: Invalid argument

addohm, 2023-10-16:
In the title you see the message I get when I try to run uStreamer.  With or without arguments, this comes up every time and the stream never starts.  This is my first time trying to set this up, or any streamer really.  F me for getting used to just duplicating images across systems to minimize work...end up falling behind in knowledge. I also have _no_ idea where logs are stored for ustreamer but I haven't been able to find anything in the typical places.  Below is the verbose start of ustreamer.

**libcamerify ustreamer --host :: -m yuyv --device-timeout=5 --buffers=3 -r 1920x1080 --log-level=3 --device=/dev/video0**
```
ERROR: ld.so: object '/usr/lib/aarch64-linux-gnu/libcamera/v4l2-compat.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
-- INFO  [23421.492      main] -- Using internal blank placeholder
-- DEBUG [23421.493      main] -- Installing SIGINT handler ...
-- DEBUG [23421.493      main] -- Installing SIGTERM handler ...
-- DEBUG [23421.493      main] -- Ignoring SIGPIPE ...
-- DEBUG [23421.493      main] -- Binding HTTP to [::]:8080 ...
-- INFO  [23421.494      main] -- Listening HTTP on [::]:8080
-- INFO  [23421.495    stream] -- Using V4L2 device: /dev/video0
-- INFO  [23421.495    stream] -- Using desired FPS: 0
-- DEBUG [23421.495    stream] -- _stream_init_loop: stream->run->stop=0
-- INFO  [23421.495      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [23421.499    stream] -- Device fd=8 opened
-- DEBUG [23421.499    stream] -- Calling ioctl(VIDIOC_QUERYCAP) ...
-- INFO  [23421.499    stream] -- Using input channel: 0
-- DEBUG [23421.499    stream] -- Using TV standard: DEFAULT
-- DEBUG [23421.500    stream] -- Calling ioctl(VIDIOC_S_FMT) ...
-- INFO  [23421.500    stream] -- Using resolution: 1920x1080
-- INFO  [23421.500    stream] -- Using pixelformat: YUYV
-- DEBUG [23421.500    stream] -- Calling ioctl(VIDIOC_G_PARM) ...
-- INFO  [23421.500    stream] -- Querying HW FPS changing is not supported
-- INFO  [23421.500    stream] -- Using IO method: MMAP
-- DEBUG [23421.500    stream] -- Calling ioctl(VIDIOC_REQBUFS) for V4L2_MEMORY_MMAP ...
-- DEBUG [23421.511      http] -- HTTP: Updating exposed frame (online=0) ...
-- VERB  [23421.512      http] -- HTTP: Exposed frame: online=0, exp_time=0.000000
-- INFO  [23421.517    stream] -- Requested 3 device buffers, got 3
-- DEBUG [23421.517    stream] -- Allocating device buffers ...
-- DEBUG [23421.517    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 0 ...
-- DEBUG [23421.517    stream] -- Mapping device buffer 0 ...
-- DEBUG [23421.518    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 1 ...
-- DEBUG [23421.518    stream] -- Mapping device buffer 1 ...
-- DEBUG [23421.518    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 2 ...
-- DEBUG [23421.518    stream] -- Mapping device buffer 2 ...
-- DEBUG [23421.518    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 0 ...
-- DEBUG [23421.518    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 1 ...
-- DEBUG [23421.519    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 2 ...
-- DEBUG [23421.519    stream] -- Device fd=8 initialized
-- DEBUG [23421.519    stream] -- Calling ioctl(VIDIOC_STREAMON) ...
-- ERROR [23421.519    stream] -- Unable to start capturing: Invalid argument
-- DEBUG [23421.521    stream] -- Releasing device buffers ...
-- DEBUG [23421.521    stream] -- Closing device ...
-- INFO  [23421.529    stream] -- Device fd=8 closed
-- INFO  [23421.529    stream] -- Sleeping 1 seconds before new stream init ...
-- DEBUG [23422.521      http] -- HTTP: Repeating exposed ...
================================================================================
```

**v4l2-ctl --all**
```
Driver Info:
        Driver name      : unicam
        Card type        : unicam
        Bus info         : platform:fe801000.csi
        Driver version   : 6.1.54
        Capabilities     : 0xa5a00001
                Video Capture
                Metadata Capture
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x25200001
                Video Capture
                Read/Write
                Streaming
                Extended Pix Format
[... 139 lines cut ...]
        [34]: 'GB14' (14-bit Bayer GBGB/RGRG)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [35]: 'pgEE' (14-bit Bayer GRGR/BGBG Packed)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [36]: 'GR14' (14-bit Bayer GRGR/BGBG)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [37]: 'pREE' (14-bit Bayer RGRG/GBGB Packed)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [38]: 'RG14' (14-bit Bayer RGRG/GBGB)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [39]: 'GREY' (8-bit Greyscale)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [40]: 'Y10P' (10-bit Greyscale (MIPI Packed))
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [41]: 'Y10 ' (10-bit Greyscale)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [42]: 'Y12P' (12-bit Greyscale (MIPI Packed))
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [43]: 'Y12 ' (12-bit Greyscale)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [44]: 'Y14P' (14-bit Greyscale (MIPI Packed))
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [45]: 'Y14 ' (14-bit Greyscale)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
```

addohm, 2023-11-12:
Dead proj or did someone take over somewhere else?

mdevaev, 2023-11-12:
The project isn't dead, I just can't test any capture device in the world.

```
ERROR: ld.so: object '/usr/lib/aarch64-linux-gnu/libcamera/v4l2-compat.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
```

This is a part of your log. Probably this is a key of the problem.

addohm, 2023-11-12:
> The project isn't dead, I just can't test any capture device in the world.
>
> ```
> ERROR: ld.so: object '/usr/lib/aarch64-linux-gnu/libcamera/v4l2-compat.so' from LD_PRELOAD cannot be preloaded (cannot [...]

I wasn't pushing you or anything, it was a legitimate question.  The issue has been around for a month without any response.  Sorry if I didn't specify but this is a Pi Camera V2 and V3 (not simultaneously).

mdevaev, 2023-11-12:
You need to figure out why libcamerafy doesn't work. I guess everything will work after that. Check also the fresh README, I added information about the V2 camera.

addohm, 2023-11-13:
> You need to figure out why libcamerafy doesn't work. I guess everything will work after that. Check also the fresh README, I added information about the V2 camera.

Will work on it some more, though not looking to go back to bullseye (where using the legacy camera option still exists).

saper-2, 2024-02-03:
Hi,

I've got bookworm on RPi too , and I stumbled upon tis same problem, my v1 camera won't work (while in alternative soft `camara-streamer` works, but....) 

Helping for me was adding `-d` (debug) to ```$ libcamerify ./ustreamer ...``` cmd line, and I got that same error:
```
ERROR: ld.so: object '/usr/lib/aarch64-linux-gnu/libcamera/v4l2-compat.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
```

A moment of my google-fu 😄 , and I found a problem source, and solution 😃 

## Problem
The problem is, that RPF moved `v4l2-compat.so` from `libcamera0` to **`libcamera-v4l2`** . 😑 

## Solution
Install: ```sudo apt install libcamera-v4l2```
This should help.

Here is post that gave me the answer: https://forums.raspberrypi.com/viewtopic.php?p=2148281#p2148281

And my cmd-line:
```
libcamerify ./ustreamer --host 0.0.0.0 --port=8080 --device=/dev/video0 -m yuyv --encoder=m2m-video --device-timeout=5 --buffers=3 --resolution=1280x720 --desired-fps=20
```

closed 2024-02-03 by mdevaev

barrelltitor, 2024-02-15:
How could the fix be applied in the docker container?

mdevaev, 2024-02-15:
By editing Dockerfile, I guess.