pikvm/ustreamer #243: Please support RK3566

Eric-168, 2023-11-15:
What happens now is when I use
sudo gst-launch-1.0 v4l2src device=/dev/video0 ! videoconvert ! capsfilter caps="video/x-raw, format=(string)YUY2" ! autovideosink
When the screen can loop out to the monitor, TC358743 can work normally, but if I use
ustreamer --device=/dev/video0 --host=0.0.0.0
will appear
root@orangepi3b:~# ustreamer --device=/dev/video0 --host=0.0.0.0
-- INFO [29990.246 main] -- Starting PiKVM uStreamer 5.37 ...
-- INFO [29990.246 main] -- Using internal blank placeholder
-- INFO [29990.246 main] -- Listening HTTP on [0.0.0.0]:8080
-- INFO [29990.247 stream] -- Using V4L2 device: /dev/video0
-- INFO [29990.247 stream] -- Using desired FPS: 0
-- INFO [29990.247 http] -- Starting HTTP eventloop ...
================================================== ==============================
-- INFO [29990.249 stream] -- Device fd=8 opened
-- ERROR [29990.249 stream] -- Video capture is not supported by device
-- INFO [29990.249 stream] -- Device fd=8 closed
-- INFO [29990.249 stream] -- Sleeping 1 seconds before new stream init ...
================================================== ==============================
As far as I know, it is because the RK3566 SOC ISP mode is different. Please adapt.

closed 2023-11-15 by mdevaev

Eric-168, 2023-11-16:
I would like to provide the orange pie in kind. Is it ok? Or Orange Pie’s ssh?

mdevaev, 2023-11-16:
No, sorry. I really don't have time to do this.

mdevaev, 2023-11-16:
Maybe it will help: https://github.com/pikvm/ustreamer/issues/97#issuecomment-1812927319