pikvm/ustreamer #244: H264: Can't open encoder device: No such file or directory

Eric-168, 2023-11-17:
I use
https://github.com/Vincent056/ustreamer/tree/rk3588-b
I successfully ran ustreamer and could obtain the video normally, but when I wanted to output h264, I encountered
root@orangepi3b:~# ustreamer -d /dev/video0 --host 0.0.0.0 --h264-sink kvmd::ustreamer::h264 --h264-sink-mode 777
-- INFO [109.183 main] -- Starting PiKVM uStreamer 5.43 ...
-- INFO [109.184 main] -- Using internal blank placeholder
-- INFO [109.184 main] -- Using H264-sink: kvmd::ustreamer::h264
-- INFO [109.184 main] -- Listening HTTP on [0.0.0.0]:8080
-- INFO [109.185 stream] -- Using V4L2 device: /dev/video0
-- INFO [109.185 stream] -- Using desired FPS: 0
-- INFO [109.185 http] -- Starting HTTP eventloop ...
-- INFO [109.185 stream] -- H264: Initializing encoder ...
-- INFO [109.191 stream] -- H264: Configuring encoder: DMA=0 ...
-- ERROR [109.191 stream] -- H264: Can't open encoder device: No such file or directory
-- ERROR [109.191 stream] -- H264: Encoder destroyed due an error (prepare)

mdevaev, 2023-11-17:
It's working only on Raspberry with hardware encoder

Eric-168, 2023-11-17:
Thank you very much, orange pi has a lower cost, I hope it can be adapted

mdevaev, 2023-11-17:
I have already told you that I officially support only Raspberry. There are hundreds of boards on ARM, each with its own surprises, and I have no desire to deal with them. I only accept patches for them, and only if they don't interfere with Raspberry support.

closed 2023-11-17 by mdevaev