pikvm/ustreamer #245: Change resolution while streaming

CursingThomas, 2023-12-06:
I want to make a feedback control loop, based on the throughput of my internet. The more it is used by other processes, the lower i want the resolution to drop; in order to save my framefrate. 

How do i dynamically adjust the resolution, is this already thought of? Can i adjust the state for example?

mdevaev, 2023-12-06:
You need to look at /state to understand which frame rate is considered normal and which frame rate the client has. Next, restart ustreamer with the required settings.

CursingThomas, 2023-12-08:
> You need to look at /state to understand which frame rate is considered normal and which frame rate the client has. Next, restart ustreamer with the required settings.

Hi, Thanks for your response. I might have stated my initial question on a bit of a time crunch, so I will try to rephrase it; since I am positive that this must be possible somehow.

According to the throughput/available bandwidth of the network, I would like to adjust the load the mediastream uses. Youtube does this, by automatically switching the stream to a lower resolution picture for example. The measuring of this throughput/bandwidth will be of course be provided by a separate application.

The goal in doing this, is to keep the fps as high as possible; since we plan to use this in a robot with a first person driving feature. That also means, that I would highly prefer this to NOT require a restart; since that would mean a reboot for a major part of the first person driving aspect. I could get around an html page restart (though not preferred), to reset the parameters of the html page output. But I would really like to keep the stream intact. 

To my knowledge, Resolution has the highest impact on the amount of bytes sent over the internet, scaling it down would mean (by my knowledge and logic): less bytes to send.
Though if you happen to know another way to do this, which might be easier or more effective, please educate me on that.

Sorry for the rather lengthy read, but I hope this provides a bit more wiggle room to discuss this further :)

mdevaev, 2023-12-08:
Well, you can try to implement what you need. There are two places that require changes:
* https://github.com/pikvm/ustreamer/blob/master/src/ustreamer/stream.c#L208, before or after v4l2 handling. If you cann `break` at this level, the stream will be restarted.
* https://github.com/pikvm/ustreamer/blob/master/src/ustreamer/device.c#L569, you also need to make some kind of function similar to the DVD timings request, but which would select the resolution you need from your list.

I'm ready to help with advice, but I won't implement this functionality and most likely won't take it upstream because it's too local a task.

CursingThomas, 2023-12-12:
Hi,

Was wondering how you might have tested network limitations on the streamer.

As far as I understand, without port forwarding; the entire system is local. LAN network.
Have you tested how network utilization affects the streamer? I want to try and prove that bad LAN network slows the stream down, though I have only managed to make it freeze. and not slow down. I did this by capping the network adapter in my system.

Have you maybe built some protection against this? Or any other pointers on how I should go about this.

mdevaev, 2023-12-13:
I'm sure network conditions affect the stream. But I haven't done any benchmarking in this direction, and I can't recommend anything.

CursingThomas, 2023-12-16:
I have made **some** advancements. I am calling the _device_apply_resolution function, which you also seem to use in the DV timing filters.
Though i am running into a bit of an error, the camera feed gets really weird. I was hoping you might provide some insights into why this is.

[image]
This is the pre-resolution change (normal, i expect this to be in the same configuration after the call; just in a different resolution)

[image]
This is after the function call, obviously not like the first.

I have checked if the resolution in the second picture actually is like the first; and isnt just correct but looks wrong. But i have verified that the resolution in the second picture should look like the first.

Any pointers? Gladly :)

mdevaev, 2023-12-16:
Considering that the picture is correct (I mean, it does not fall apart into artifacts), then you get exactly this image from the signal source. Maybe you chose the wrong resolution?

CursingThomas, 2023-12-17:
> Considering that the picture is correct (I mean, it does not fall apart into artifacts), then you get exactly this image from the signal source. Maybe you chose the wrong resolution?

Hi, I was thinking that too; but no. I verified that by running the stream with the same resolution I dynamically switched two (aka the resolution from picture 2). But to no avail. 

If I pick a resolution small enough whereas the camera only gives me one image (stereo camera, so big resolution means output of both; small resolution means output of one), the image is not centered. To get my hand in the middle of the frame i have to offset it to the side. Maybe that provides a clue as to where the problem lies.

Something is messing up something, but I would not know where; I hoped you might.

mdevaev, 2023-12-17:
The only idea I have is reopen device file :/

CursingThomas, 2023-12-18:
> The only idea I have is reopen device file :/

Ok, will look into that tomorrow. Thanks so far

CursingThomas, 2023-12-19:
Ok i am getting closer i feel like, i am using US_Device_open(stream->dev); to reopen the device file; though i get the error that i cannot set an input channel; probably since it happens so fast the program still thinks its in use. Do you know a workaround? Device_close() first does not work.

mdevaev, 2023-12-19:
Just comment code with choosing device input in device.c

CursingThomas, 2023-12-22:
> Just comment code with choosing device input in device.c

Ok, i have investigated and as of now i have moved on from trying to implement a changing resolution. I came across another bug where the semi-changing of resolution like shown only worked with the YUYV algorithm.

Instead i am looking into variable quality from the encoder. I have noticed though that with MJPEG the encoder automatically switches to a hardware, did you do this on purpose given that no quality scaling would be neccesairy; or is this a restriction somewhere down the pipeline for whatever reason?

mdevaev, 2023-12-24:
This is the input format selection. There is no point in re-encoding the stream if it is already in MJPEG. You can choose any RAW format and use the CPU encoder, and quickly change the quality. By the way, not all hardware decoders are capable of changing the quality.

CursingThomas, 2023-12-24:
> This is the input format selection. There is no point in re-encoding the stream if it is already in MJPEG. You can choose any RAW format and use the CPU encoder, and quickly change the quality. By the way, not all hardware encoders are capable of changing the quality.

I think you might be understanding me wrong, my camera can use two formats; yuyv and mjpeg. yuyv gives horrible fps and is not suited for my application where i need the highest fps possible. 

The hardware encoder you added does not have a feature to set/select quality; since i believe it just copies the file it gets from the camera 1:1. Atleast it says that in the comments.

Why is this? The CPU encoder supports quality, why doesnt the HW encoder?

I want to know this, since i cannot select any raw format, i can only select yuyv and mjpeg.

mdevaev, 2024-01-10:
Sorry, I forgot to answer.

> Why is this? The CPU encoder supports quality, why doesnt the HW encoder?

The hardware encoder works inside your camera. Some support changing the quality settings, but most do not.

mdevaev, 2024-01-28:
Hello! How it's going?

mdevaev, 2024-02-03:
Sup?

mdevaev, 2024-02-05:
Okay, it's probably not relevant anymore. Let me know if you need anything else.

closed 2024-02-05 by mdevaev