pikvm/ustreamer #246: Mjpeg Streaming stops randomly in Edge and Chrome in OneCloud board

mofeng-git, 2023-12-10:
There is a problem in is s805 development board using USB HDMI capture card, ustreamer using mjpeg format（yuyv format is normal） will appear in Edge, Chrome browser frequent video screen interruptions black screen phenomenon (a few minutes of great probability of black screen), can be reset by resetting the video stream (in the case of pikvm) or refresh the web page temporary briefly restored to the screen, the two sides of the no error report or abnormal logs. But on Firefox, this is all normal.

Check ustreamer, mjpeg format capture encoding logs are no error, and no network reasons (after the black screen will continue to transmit video data, bandwidth consumption remains unchanged)
Just like https://github.com/pikvm/ustreamer/issues/217

Lowering the CPU frequency of OneCloud to 600MHz has a miraculous effect on this problem, which can greatly delay the occurrence of the black screen phenomenon time (half an hour later, there is a probability of black screen), but the problem has not been completely solved.So I think this question may be related to ustreamer.
`cpufreq-set -d 400MHz -u 600MHz && systemctl restart kvmd kvmd-vnc `

So I'd like to seek for help and or idea for continued testing。

To serve as a reference，everything is fine in Orangepi zero (no black screen). the Orangepi zero CPU is 1.2G and the OneCloud board CPU is 1.54G.
Test environment (abnormal) : OneCloud (s805 1G+8G) Armbian Jammy (armv7 kernel 5.15.82/5.15.85/5.10.149/6.1.12/6.1.63) (host/docker) ustreamer (1.0/2.1/3.0/4.0/5.37/5.45 )
Test environment (normal): Orangepi zero Armbian Jammy ( armv7 kernel 5.15.81) host ustreamer 5.37
[image]
[image]

mdevaev, 2023-12-10:
MJPEG is encoded on the side of the capture device, so it is probably the problem, given that the bandwidth does not change with a black screen. I can't say what the reason is, but it's probably a hardware problem. ustreamer doesn't do anything with the video, it just transmits what it receives.

closed 2023-12-14 by mdevaev

mofeng-git, 2023-12-17:
Thanks for the reply. I guess the reason was that the system default on demand CPU frequency setting was making ustreamer run on different CPU frequencies. I solved the problem completely by setting a fixed CPU frequency.