pikvm/ustreamer #247: ustreamer-master make error

clhchan, 2023-12-23:
Hello author, I encountered these error messages while compiling with make. What could be the issue? It's the Bookworm version for Raspberry Pi5

pi@raspberrypi:~/ustreamer-master$ make
make apps
make[1]: Entering directory '/home/pi/ustreamer-master'
make -C src
make[2]: Entering directory '/home/pi/ustreamer-master/src'
-- CC ustreamer/http/server.c
-- CC ustreamer/http/static.c
-- CC ustreamer/http/unix.c
-- CC ustreamer/http/uri.c
-- CC ustreamer/m2m.c
-- CC ustreamer/main.c
-- CC ustreamer/options.c
-- CC ustreamer/stream.c
-- CC ustreamer/workers.c
== LD ustreamer.bin
/usr/bin/ld: build/ustreamer/http/server.o: in function `_http_callback_stream_error':
server.c:(.text+0x1f30): undefined reference to `us_bufferevent_format_reason'
/usr/bin/ld: build/ustreamer/http/server.o: in function `_http_callback_static.part.0':
server.c:(.text+0x40f0): undefined reference to `us_guess_mime_type'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:89: ustreamer.bin] Error 1
make[2]: Leaving directory '/home/pi/ustreamer-master/src'
make[1]: *** [Makefile:38: apps] Error 2
make[1]: Leaving directory '/home/pi/ustreamer-master'
make: *** [Makefile:28: all] Error 2

mdevaev, 2024-01-11:
I can't reproduce it. Try `make clean-all` and `make` it again.

mdevaev, 2024-01-28:
No answer from user.

closed 2024-01-28 by mdevaev