pikvm/ustreamer #248: Raspberry Pi 5 with Camera Module V3

Samueleitor6, 2023-12-29:
Hi there! I've been using streamer with Raspberry Pi 4 with raspbian Bulleye, but when y switched to Raspberry Pi 5 with Raspbian Bookworm I've been trying to start stream but I can't. This is the log:

-- INFO  [55007.890    stream] -- Device fd=8 opened
-- INFO  [55007.890    stream] -- Using capture type: single-planar
-- INFO  [55007.890    stream] -- Using input channel: 0
-- INFO  [55007.890    stream] -- Using resolution: 1920x1080
-- INFO  [55007.890    stream] -- Using format: YUYV
-- INFO  [55007.890    stream] -- Querying HW FPS changing is not supported
-- INFO  [55007.890    stream] -- Using IO method: MMAP
-- INFO  [55007.893    stream] -- Requested 2 device buffers, got 3
-- ERROR [55007.893    stream] -- Can't start capturing: Invalid argument
-- INFO  [55007.894    stream] -- Device fd=8 closed
-- INFO  [55007.894    stream] -- Sleeping 1 seconds before new stream init ...

If somebody knows a way to make it work with this hardware and software that tells me here.

Thank you very much in advance!

Crocodile94, 2024-01-29:
Same hardware with HQ camera and i got this error too.

mdevaev, 2024-02-03:
Hello. Please install `libcamera-tools` and `libcamera-v4l2` and try `libcamerify ustreamer ...`.

Samueleitor6, 2024-02-03:
Thank you so much!! That solve my issue, it works perfectly now!

closed 2024-02-03 by Samueleitor6

mdevaev, 2024-02-03:
:ok_hand: