pikvm/ustreamer #252: ustreamer won't start because unsupported YVYU format

bschwand, 2024-01-11:
trying a small cam (from a laptop originally), I tested it before with a cam viewer so I know it works

 sudo systemctl status ustreamer@1.service
......
Jan 11 20:04:31 orangepione ustreamer[2439]: -- ERROR [9999.383    stream] -- Could not obtain the requested pixelformat=YUYV; driver gave us unsupported
Jan 11 20:04:31 orangepione ustreamer[2439]: -- ERROR [9999.383    stream] -- Unsupported pixelformat=YVYU (fourcc)

bruno@orangepione:/home/bruno$ v4l2-ctl -d /dev/video1 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YVYU' (YVYU 4:2:2)
                Size: Discrete 320x240
                Size: Discrete 640x480

any way this format could be supported ?

mdevaev, 2024-01-11:
Hello. I've made a quick implementation of this. Please compile ustreamer from branch `yvyu` and try it.

mdevaev, 2024-01-12:
Sup?

bschwand, 2024-01-13:
sorry I missed the notification email and was just testing pretty much the same changes
it works, in the sense that it starts streaming, but the colors are all wrong. looks red blueish. gonna see if swapping things around fixes the colors...

bschwand, 2024-01-13:
yes, I know ;-)
I should have just fixed it myself and sent a pull request but got derailed by other projects. We all know how quick fixes turn into a week long research.

So, with the logical byteswpap given the format (like your yvyu branch) 
what it is looking at:
[image]

what it actually looks like:
[image]

I am trying to debate wether there is something strange in the pixel ordering and colorspace conversion, or if it's just a shitty camera

mdevaev, 2024-02-02:
For some reason, I missed your message. Okay, I'm merging this, thanks for the testing!

mdevaev, 2024-02-02:
Ok, merged. I will make a new release soon.

closed 2024-02-02 by mdevaev