pikvm/ustreamer #256: Mainloop select() error: Success

Assassins88, 2024-02-09:
I get the following error:

```
- INFO  [1893.164    stream] -- Device fd=8 opened
-- INFO  [1893.164    stream] -- Using capture type: single-planar
-- INFO  [1893.164    stream] -- Using input channel: 0
-- INFO  [1893.190    stream] -- Using resolution: 640x480
-- INFO  [1893.190    stream] -- Using format: YUYV
-- INFO  [1893.216    stream] -- Using HW FPS: 0 -> 30 (coerced)
-- INFO  [1893.216    stream] -- Using IO method: MMAP
-- INFO  [1893.217    stream] -- Requested 5 device buffers, got 5
-- INFO  [1893.233    stream] -- Capturing started
-- INFO  [1893.233    stream] -- Using JPEG quality: 80%
-- INFO  [1893.233    stream] -- Creating pool JPEG with 4 workers ...
-- INFO  [1893.233    stream] -- Capturing ...
-- ERROR [1894.235    stream] -- Mainloop select() error: Success
-- INFO  [1894.235    stream] -- Destroying workers pool JPEG ...
-- INFO  [1894.278    stream] -- Capturing stopped
-- INFO  [1894.282    stream] -- Device fd=8 closed
```

This advertisement repeats itself over and over again.

I run the standart command without parameters.

The system runs in a Proxmox VM without a GPU so i cant use hw encoding and MJPG...

mdevaev, 2024-02-09:
Which capture device do you use?

Assassins88, 2024-02-09:
> Which capture device do you use?

I'm sorry, but what do you mean exactly?

mdevaev, 2024-02-09:
Tell me how you launched ustreamer, which video capture device (camera?) you use it. More information, I don't understand what's going on.

Assassins88, 2024-02-09:
> Tell me how you launched ustreamer, which video capture device (camera?) you use it. More information, I don't understand what's going on.

I launched it with different commands:

```
./ustreamer

./ustreamer --device=/dev/video0 --host=0.0.0.0 --port=80 -m YVYU

./ustreamer --device=/dev/video0 --host=0.0.0.0 --port=80 -m mjpeg
```

with the mjpeg parameter comes a error that my device dosnt support hw encoding...

my cam supports the following formats:
```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 544x288
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 432x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 320x176
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 752x416
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 800x448
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 864x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 960x544
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 960x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1024x576
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1184x656
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 544x288
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 432x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 320x176
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 752x416
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 800x448
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 864x480
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 960x544
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 1024x576
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 960x720
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1184x656
			Interval: Discrete 0.100s (10.000 fps)

```

The cam is a cheap Hama webcam but it works with MJPG-Streamer.

mdevaev, 2024-02-09:
> with the mjpeg parameter comes a error that my device dosnt support hw encoding...

Which message exactly?

Assassins88, 2024-02-09:
> > with the mjpeg parameter comes a error that my device dosnt support hw encoding...
> 
> Which message exactly?

 Device doesn't support setting of HW encoding quality parameters

mdevaev, 2024-02-10:
Try `--io-method=USERPTR`

Assassins88, 2024-02-10:
> Try `--io-method=USERPTR`

First came the error message above and then the following:

```
-- INFO  [69092.741    stream] -- Device fd=0 opened
-- INFO  [69092.741    stream] -- Using capture type: single-planar
-- INFO  [69092.741    stream] -- Using input channel: 0
-- INFO  [69092.768    stream] -- Using resolution: 640x480
-- INFO  [69092.768    stream] -- Using format: YUYV
-- INFO  [69092.794    stream] -- Using HW FPS: 0 -> 30 (coerced)
-- INFO  [69092.794    stream] -- Using IO method: USERPTR
-- INFO  [69092.794    stream] -- Requested 5 device buffers, got 5
-- INFO  [69092.811    stream] -- Capturing started
-- INFO  [69092.811    stream] -- Using JPEG quality: 80%
-- INFO  [69092.811    stream] -- Creating pool JPEG with 4 workers ...
-- INFO  [69092.812    stream] -- Capturing ...
-- ERROR [69093.813    stream] -- Mainloop select() error: Bad file descriptor
-- INFO  [69093.813    stream] -- Destroying workers pool JPEG ...
-- INFO  [69093.857    stream] -- Capturing stopped
-- ERROR [69093.860    stream] -- Can't close device fd=0: Bad file descriptor

```

mdevaev, 2024-02-10:
Then I don't know what the reason may be, sorry. I don't see any visible difference with mjpg-streamer with select() and I don't have cameras with the same glitch.

SageStarCodes, 2024-04-28:
I get the same error using PS3 Eye camera
```
ellie@pi3:~$ ustreamer
-- INFO  [10709.277      main] -- Using internal blank placeholder
-- INFO  [10709.279      main] -- Listening HTTP on [127.0.0.1]:8080
-- INFO  [10709.280    stream] -- Using V4L2 device: /dev/video0
-- INFO  [10709.280    stream] -- Using desired FPS: 0
-- INFO  [10709.280      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [10709.282    stream] -- Device fd=8 opened
-- INFO  [10709.282    stream] -- Using input channel: 0
-- INFO  [10709.282    stream] -- Using resolution: 640x480
-- INFO  [10709.282    stream] -- Using pixelformat: YUYV
-- INFO  [10709.282    stream] -- Using HW FPS: 0 -> 255 (coerced)
-- INFO  [10709.282    stream] -- Using IO method: MMAP
-- INFO  [10709.285    stream] -- Requested 5 device buffers, got 5
-- INFO  [10710.175    stream] -- Capturing started
-- INFO  [10710.175    stream] -- Using JPEG quality: 80%
-- INFO  [10710.175    stream] -- Creating pool JPEG with 4 workers ...
-- INFO  [10710.178    stream] -- Capturing ...
-- ERROR [10711.179    stream] -- Mainloop select() error: Success
-- INFO  [10711.179    stream] -- Destroying workers pool JPEG ...
-- INFO  [10711.184    stream] -- Capturing stopped
-- INFO  [10711.186    stream] -- Device fd=8 closed
================================================================================
-- INFO  [10711.189    stream] -- Device fd=8 opened
-- INFO  [10711.190    stream] -- Using input channel: 0
-- INFO  [10711.190    stream] -- Using resolution: 640x480
-- INFO  [10711.190    stream] -- Using pixelformat: YUYV
-- INFO  [10711.190    stream] -- Using HW FPS: 0 -> 255 (coerced)
-- INFO  [10711.190    stream] -- Using IO method: MMAP
-- INFO  [10711.192    stream] -- Requested 5 device buffers, got 5
^C-- INFO  [10711.585      main] -- ===== Stopping by SIGINT =====
-- INFO  [10711.585      http] -- HTTP eventloop stopped
-- INFO  [10712.098    stream] -- Capturing started
-- INFO  [10712.098    stream] -- Using JPEG quality: 80%
-- INFO  [10712.098    stream] -- Creating pool JPEG with 4 workers ...
-- INFO  [10712.100    stream] -- Capturing ...
-- INFO  [10712.100    stream] -- Destroying workers pool JPEG ...
-- INFO  [10712.103    stream] -- Capturing stopped
-- INFO  [10712.105    stream] -- Device fd=8 closed
-- INFO  [10712.107      main] -- Bye-bye
```

```
ellie@pi3:~$ ustreamer --device=/dev/video0 --host=0.0.0.0 -c NOOP
-- INFO  [10828.772      main] -- Using internal blank placeholder
-- INFO  [10828.773      main] -- Listening HTTP on [0.0.0.0]:8080
-- INFO  [10828.774    stream] -- Using V4L2 device: /dev/video0
-- INFO  [10828.774    stream] -- Using desired FPS: 0
-- INFO  [10828.774      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [10828.776    stream] -- Device fd=8 opened
-- INFO  [10828.776    stream] -- Using input channel: 0
-- INFO  [10828.776    stream] -- Using resolution: 640x480
-- INFO  [10828.777    stream] -- Using pixelformat: YUYV
-- INFO  [10828.777    stream] -- Using HW FPS: 0 -> 255 (coerced)
-- INFO  [10828.777    stream] -- Using IO method: MMAP
-- INFO  [10828.779    stream] -- Requested 5 device buffers, got 5
-- INFO  [10829.689    stream] -- Capturing started
-- INFO  [10829.689    stream] -- Using JPEG NOOP encoder
-- INFO  [10829.689    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [10829.690    stream] -- Capturing ...
-- ERROR [10830.691    stream] -- Mainloop select() error: Success
-- INFO  [10830.691    stream] -- Destroying workers pool JPEG ...
-- INFO  [10830.694    stream] -- Capturing stopped
-- INFO  [10830.696    stream] -- Device fd=8 closed
================================================================================
-- INFO  [10830.700    stream] -- Device fd=8 opened
-- INFO  [10830.700    stream] -- Using input channel: 0
[... 108 lines cut ...]
Package: ustreamer
Version: 4.9-1build2
Priority: optional
Section: universe/video
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Sam Reed <sam@reedyboy.net>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 420 kB
Depends: libbsd0 (>= 0.6.0), libc6 (>= 2.38), libevent-2.1-7 (>= 2.1.8-stable), libevent-pthreads-2.1-7 (>= 2.1.8-stable), libjpeg8 (>= 8c)
Homepage: https://pikvm.org/
Download-Size: 118 kB
APT-Manual-Installed: yes
APT-Sources: http://ports.ubuntu.com/ubuntu-ports mantic/universe arm64 Packages
Description: Lightweight and fast MJPG-HTTP streamer
 µStreamer is a lightweight and very quick server to stream MJPG video
 from any V4L2 device to the network. All new browsers have native
 support of this video format, as well as most video players such as
 mplayer, VLC etc. µStreamer is a part of the Pi-KVM project designed to
 stream VGA and HDMI screencast hardware data with the highest resolution
 and FPS possible.
```

If any other logs would be useful, let me know, I can also test stuff if you would like me to do so :)
(dmesg and journal are both clean)

mdevaev, 2024-05-01:
You're using a very old ustreamer release, please build it from git and test again.

closed 2025-11-17 by mdevaev