pikvm/ustreamer #258: "ERROR Got unexpected writing event, seems device was disconnected" Raspberry Pi5 with Camera Module V2

lgeitner, 2024-02-13:
I am using a Raspberry PI 5 8GB with an official Raspberry PI Camera Module V2

I used Raspberry PI OS 64 Bit Bookworm Image from 12-05-2023

pi@octoprinttest:~ $ uname -a
Linux octoprinttest 6.1.65-v8-16k+ #1703 SMP PREEMPT Tue Dec  5 16:28:46 GMT 2023 aarch64 GNU/Linux

In my /boot/firmware/config.txt I have added camera_auto_detect=1

After the PI booted up I issued the following commands

sudo apt update
sudo apt full-upgrade -y
sudo reboot
sudo apt install libcamera-tools libcamera-v4l2 ustreamer -y
sudo reboot
sudo rpi-update d16727d #downgrade kernel to 6.1.65 so V2 camera will work
sudo reboot
rpicam-hello #This displays video from the camera.  So I know I have the camera connected correctly!
sudo raspi-config nonint do_legacy 0 #enable legacy camera mode in raspi-config
sudo modprobe bcm2835-v4l2

when I try

libcamerify ustreamer --host :: --encoder=m2m-image # per the example

I receive the error "Unknown encoder type: m2m-image; available: CPU, HW, NOOP" 

which is probably because the pi 5 does not have hardware encoders????

Whether I try any of the following below: I receive the error -- ERROR [241.717    stream] -- Got unexpected writing event, seems device was disconnected

libcamerify ustreamer --host :: --encoder=CPU
libcamerify ustreamer --host :: --encoder=HW
libcamerify ustreamer --host :: --encoder=NOOP

I have googled the above error and found no error like that!

Any ideas what I am doing wrong?
Could this be a libcamerify issue?

Thanks,
Lance

mdevaev, 2024-02-13:
Hello. Don't use ustreamer from the repo, it's very old there. Please build it from the source, your issue is already fixed.

lgeitner, 2024-02-13:
Thank You, that fixed that error.

I am also receiving the below error.  Can this error be ignored?

[1:16:55.834548056] [7568] ERROR IPAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
[1:16:55.834770241] [7568] ERROR IPAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info

Thanks,
Lance

mdevaev, 2024-02-14:
If everything is working, I suppose you can ignore it :) This is from libcamera, I don't know what it means.

lgeitner, 2024-02-14:
Ok.  Thank You.

mdevaev, 2024-02-14:
:ok_hand:

closed 2024-02-14 by mdevaev