pikvm/ustreamer #260: Unable to use GPU-accelerated encoding on Raspberry PI 5 (--encoder=M2M-IMAGE or --encoder=M2M-VIDEO)

lgeitner, 2024-02-16:
Official Raspberry PI 5 
Official Raspberry PI Camera Module 2
Official Raspberry PI Power Supply (27 Watt)
Official Raspberry PI Camera Cable 300MM

Used Raspberry PI OS 64 Bit Lite

when running the following command

pi@pitest:~/ustreamer $ libcamerify ./ustreamer -d /dev/video0 -s 0.0.0.0 -m YUYV -r 1280x720 -f 40 -p 8001 --encoder=M2M-IMAGE --device-timeout 8 --device-error-delay 8

I receive the below error:

-- INFO  [9470.046    stream] -- Capturing ...
-- INFO  [9470.403      jw-0] -- JPEG-0: Configuring encoder: DMA=0 ...
-- ERROR [9470.403      jw-0] -- JPEG-0: Can't set option V4L2_CID_JPEG_COMPRESSION_QUALITY: Inappropriate ioctl for device
-- ERROR [9470.403      jw-0] -- JPEG-0: Encoder destroyed due an error (prepare)
-- ERROR [9470.403      jw-0] -- Compression failed: worker=jw-0, buffer=0
-- ERROR [9470.403      jw-0] -- Error while compressing buffer, falling back to CPU
-- INFO  [9470.403    stream] -- Destroying workers pool JPEG ...

This does work using the CPU however there is between 40%-50% CPU Utilization!!!!!!!!!

Is there any way to reduce CPU Utilization?

 I have tried lowering the resolution and frame rate, but the CPU Utilization is still around 40-50%

according to the help file I should be able to specify --encoder=M2M-IMAGE or --encoder=M2M-VIDEO to use GPU accelerated encoding.

if I use the option --encoder=NOOP I do not see any video at all.

**I realize that --encoder=HW is not going to work on a pi 5 as the PI 5 has no hardware encoder!**

Thank You,
Lance
  
  -c|--encoder <type>  ───────────────── Use specified encoder. It may affect the number of workers.
               Available:
                       * CPU  ──────── Software MJPEG encoding (default);
                       * HW  ───────── Use pre-encoded MJPEG frames directly from camera hardware;
                       * M2M-VIDEO  ── GPU-accelerated MJPEG encoding using V4L2 M2M video interface;
                       * M2M-IMAGE  ── GPU-accelerated JPEG encoding using V4L2 M2M image interface;
                       * NOOP  ─────── Don't compress MJPEG stream (do nothing).

pi@pitest:~/ustreamer $ uname -a
Linux pitest 6.1.65-v8-16k+ #1703 SMP PREEMPT Tue Dec  5 16:28:46 GMT 2023 aarch64 GNU/Linux

pi@pitest:~/ustreamer $ ./ustreamer -v
5.50

pi@pitest:~/ustreamer $ v4l2-ctl --all -d /dev/video0
Driver Info:
        Driver name      : rp1-cfe
        Card type        : rp1-cfe
        Bus info         : platform:1f00128000.csi
        Driver version   : 6.1.65
        Capabilities     : 0xaca00001
                Video Capture
                Metadata Capture
                Metadata Output
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x24200001
                Video Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : rp1-cfe
        Model            : rp1-cfe
        Serial           :
        Bus info         : platform:1f00128000.csi
        Media version    : 6.1.65
        Hardware revision: 0x00114666 (1132134)
        Driver version   : 6.1.65
Interface Info:
        ID               : 0x03000015
        Type             : V4L Video
Entity Info:
        ID               : 0x00000013 (19)
        Name             : rp1-cfe-csi2_ch0
        Function         : V4L2 I/O
        Pad 0x01000014   : 0: Sink, Must Connect
          Link 0x02000035: from remote pad 0x1000006 of entity 'csi2' (Video Interface Bridge): Data
Priority: 2
Video input : 0 (rp1-cfe-csi2_ch0: ok)
Format Video Capture:
        Width/Height      : 640/480
        Pixel Format      : 'pRAA' (10-bit Bayer RGRG/GBGB Packed)
        Field             : None
        Bytes per Line    : 800
        Size Image        : 384000
        Colorspace        : Raw
        Transfer Function : None
        YCbCr/HSV Encoding: ITU-R 601
        Quantization      : Full Range
        Flags             :

mdevaev, 2024-02-16:
As you have already found out, the Raspberry Pi 5 does not have a hardware GPU encoder at all. There's nothing we can do about it. The CPU is just doing its job.

mdevaev, 2024-02-17:
If you have the appropriate V4L2 driver, then this can work. But I haven't tried it in practice and I don't know if it's possible.

lgeitner, 2024-02-17:
Ok.  Thank You.

mdevaev, 2024-02-17:
:ok_hand:

closed 2024-02-17 by mdevaev