pikvm/ustreamer #261: ustreamer outputs a large amount of logs: `V4L2 error: grabbed device buffer=0 is already used`

thenbe, 2024-02-17:
## Issue

Hi. Sometimes, ustreamer enters a state that produces a massive amount of logs (approx. 10,779 lines of logs per second!). The error that keeps repeating is: `V4L2 error: grabbed device buffer=0 is already used`.

<details>
  <summary>Lots of repetitive logs</summary>

```
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.712    stream] -- Device fd=8 opened
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.712    stream] -- Using capture type: single-planar
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.712    stream] -- Using input channel: 0
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.712    stream] -- Using resolution: 1920x1080
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.712    stream] -- Using format: MJPEG
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.712    stream] -- Changing HW FPS is not supported
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- ERROR [43.712    stream] -- Device doesn't support setting of HW encoding quality parameters
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.712    stream] -- Using IO method: MMAP
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.712    stream] -- Requested 1 device buffers, got 1
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.713    stream] -- Capturing started
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.713    stream] -- Using JPEG quality: encoder default
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.713    stream] -- Creating pool JPEG with 1 workers ...
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.713    stream] -- Capturing ...
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- ERROR [43.713    stream] -- V4L2 error: grabbed device buffer=0 is already used
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.713    stream] -- Destroying workers pool JPEG ...
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.713    stream] -- Capturing stopped
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.714    stream] -- Device fd=8 closed
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: ================================================================================
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.714    stream] -- Device fd=8 opened
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.714    stream] -- Using capture type: single-planar
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.714    stream] -- Using input channel: 0
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.714    stream] -- Using resolution: 1920x1080
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.714    stream] -- Using format: MJPEG
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.714    stream] -- Changing HW FPS is not supported
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- ERROR [43.714    stream] -- Device doesn't support setting of HW encoding quality parameters
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.714    stream] -- Using IO method: MMAP
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.714    stream] -- Requested 1 device buffers, got 1
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.715    stream] -- Capturing started
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.715    stream] -- Using JPEG quality: encoder default
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.715    stream] -- Creating pool JPEG with 1 workers ...
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.716    stream] -- Capturing ...
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- ERROR [43.716    stream] -- V4L2 error: grabbed device buffer=0 is already used
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.716    stream] -- Destroying workers pool JPEG ...
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.716    stream] -- Capturing stopped
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.717    stream] -- Device fd=8 closed
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: ================================================================================
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.717    stream] -- Device fd=8 opened
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.717    stream] -- Using capture type: single-planar
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.717    stream] -- Using input channel: 0
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.717    stream] -- Using resolution: 1920x1080
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.717    stream] -- Using format: MJPEG
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.717    stream] -- Changing HW FPS is not supported
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- ERROR [43.717    stream] -- Device doesn't support setting of HW encoding quality parameters
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.717    stream] -- Using IO method: MMAP
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.717    stream] -- Requested 1 device buffers, got 1
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.718    stream] -- Capturing started
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.718    stream] -- Using JPEG quality: encoder default
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.718    stream] -- Creating pool JPEG with 1 workers ...
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.718    stream] -- Capturing ...
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- ERROR [43.718    stream] -- V4L2 error: grabbed device buffer=0 is already used
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.718    stream] -- Destroying workers pool JPEG ...
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.720    stream] -- Capturing stopped
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.721    stream] -- Device fd=8 closed
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: ================================================================================
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.721    stream] -- Device fd=8 opened
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.721    stream] -- Using capture type: single-planar
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.721    stream] -- Using input channel: 0
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.722    stream] -- Using resolution: 1920x1080
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.722    stream] -- Using format: MJPEG
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.722    stream] -- Changing HW FPS is not supported
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- ERROR [43.722    stream] -- Device doesn't support setting of HW encoding quality parameters
Sat 2024-02-17 14:24:21 +03 seven-mangoes ustreamer.service[1365]: -- INFO  [43.722    stream] -- Using IO method: MMAP
[... 152 lines cut ...]
Sat 2024-02-17 14:45:44 +03 seven-mangoes ustreamer.service[1207]: -- INFO  [136.824    stream] -- Capturing started
[... cut ...]
Sat 2024-02-17 14:45:44 +03 seven-mangoes ustreamer.service[1207]: -- INFO  [136.824    stream] -- Capturing ...
```

</details>

## Other

I only added the flags `--buffers=1 --workers=1` as a wild attempt, but it does not seem to have an effect (not sure if it even should tbh).

## Versions

```
 - ustreamer: `v5.50`
 - hardware: Raspberry Pi 400
 - system: `"aarch64-linux"`
 - host os: `Linux 6.1.63, NixOS, 24.05 (Uakari), 24.05pre-git`
 - multi-user?: `yes`
 - sandbox: `yes`
 - version: `nix-env (Nix) 2.18.1`
 - channels(root): `"nixos-23.11pre-git"`
 - nixpkgs: `/etc/nix/inputs/nixpkgs`
```

mdevaev, 2024-02-17:
Hi.

This error means that the V4L2 video capture device driver is not working properly. The driver must provide one of the buffers to the device, and the application (ustreamer) can request several one after the other without returning the previous ones.

In the case of --workers=1, ustream also checks that the buffer received from the driver was not received twice.

This code is well-debugged and works because we use it for HDMI video capture, but I checked everything again. I think you should inform the driver/kernel developers about this problem.

thenbe, 2024-02-17:
In that case I wonder if the issue might be with how I start my virtual device:

```
v4l2loopback-ctl add -x 1 /dev/video0
v4l2loopback-ctl set-caps /dev/video0 'MJPG:1920x1080@30/1'
v4l2loopback-ctl set-fps /dev/video0 30
```

I've come across three alternate ways to go accomplish this (e.g. [`v4l2-ctl --help`](https://www.mankier.com/1/v4l2-ctl#) vs [`modprobe v4l2loopback`](https://github.com/umlaeute/v4l2loopback?tab=readme-ov-file#v4l2loopback---a-kernel-module-to-create-v4l2-loopback-devices) vs [`v4l2loopback-ctl`](https://www.mankier.com/1/v4l2loopback-ctl)) so I'll try to see if using the others makes a difference here.

Out of curiosity, are any of these how you typically initialize your virtual devices as well, or do you use something else?

---

> inform the driver/kernel developers

Would you happen to know if this (https://github.com/umlaeute/v4l2loopback) is the relevant repo to do that?

Thanks for checking up on the issue, I'm going to close this now as the source of the bug seems to lie elsewhere.

closed 2024-02-17 by thenbe

mdevaev, 2024-02-17:
> Would you happen to know if this (https://github.com/umlaeute/v4l2loopback) is the relevant repo to do that?

Yes, it is. Yes, this is definitely a v4l2loopback issue, I have already been informed about similar issues before.

thenbe, 2024-02-18:
After debugging some more, I think this behavior may have been caused as a result of using v4l2loopback's [dynamic device management](https://github.com/umlaeute/v4l2loopback?tab=readme-ov-file#dynamic-device-management). The issue does not seem to happen when I use `modprobe` to declare and customize the virtual device.