pikvm/ustreamer #263: 32 bit doesn't build anymore

mryel00, 2024-03-02:
Commit https://github.com/pikvm/ustreamer/commit/54b221aabdd75a53304e54b5932f76e14c6f84b5 added `atomic_uint_least64_t` inside `src/ustreamer/stream.h`. This then got changed to `atomic_ullong` in commit https://github.com/pikvm/ustreamer/commit/e2f4c193e353c831f39e199e3b240fb636d5fe5d.
Due to this change 32 bit systems can no longer build ustreamer, because `atomic_ullong` is a 64 bit data type. Using `atomic_ulong` would fix this issue.
Is there any specific reason to why this 64 bit data type was used instead of the 32 bit one? Or was it just an oversight?

mdevaev, 2024-03-02:
I'll fix it, thanks. It's strange that it still works on my 32 bit ARM.

mryel00, 2024-03-02:
Thanks for fixing it.
Just fyi, it can be reproduced on a Raspberry Pi 32 bit Bullseye image using `make `. I didn't test Bookworm.

mdevaev, 2024-03-02:
~~I'm using arch btw~~ :)

mryel00, 2024-03-02:
Just for the record, as I never mentioned the error:
```
Build 'ustreamer' using 2 Cores ...
make apps
make[1]: Entering directory '/home/mryel/crowsnest/bin/ustreamer'
make -C src
make[2]: Entering directory '/home/mryel/crowsnest/bin/ustreamer/src'
-- CC libs/base64.c
-- CC libs/device.c
-- CC libs/frame.c
-- CC libs/frametext.c
-- CC libs/frametext_font.c
-- CC libs/logging.c
-- CC libs/memsink.c
-- CC libs/options.c
-- CC libs/queue.c
-- CC libs/ring.c
-- CC libs/unjpeg.c
-- CC ustreamer/blank.c
-- CC ustreamer/data/favicon_ico.c
-- CC ustreamer/data/index_html.c
-- CC ustreamer/encoder.c
-- CC ustreamer/encoders/cpu/encoder.c
-- CC ustreamer/encoders/hw/encoder.c
-- CC ustreamer/h264.c
-- CC ustreamer/http/bev.c
-- CC ustreamer/http/mime.c
-- CC ustreamer/http/path.c
-- CC ustreamer/http/server.c
-- CC ustreamer/http/static.c
-- CC ustreamer/http/unix.c
-- CC ustreamer/http/uri.c
-- CC ustreamer/m2m.c
-- CC ustreamer/main.c
-- CC ustreamer/options.c
-- CC ustreamer/stream.c
-- CC ustreamer/workers.c
-- CC dump/file.c
-- CC dump/main.c
== LD ustreamer.bin
/usr/bin/ld: build/ustreamer/http/server.o: in function `_http_preprocess_request':
server.c:(.text+0x768): undefined reference to `__atomic_store_8'
/usr/bin/ld: build/ustreamer/stream.o: in function `_stream_check_suicide':
stream.c:(.text+0x1b5c): undefined reference to `__atomic_load_8'
/usr/bin/ld: stream.c:(.text+0x1d00): undefined reference to `__atomic_store_8'
/usr/bin/ld: build/ustreamer/stream.o: in function `us_stream_init':
stream.c:(.text+0x28ac): undefined reference to `__atomic_store_8'
/usr/bin/ld: build/ustreamer/stream.o: in function `us_stream_loop':
stream.c:(.text+0x2c40): undefined reference to `__atomic_store_8'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:109: ustreamer.bin] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/mryel/crowsnest/bin/ustreamer/src'
make[1]: *** [Makefile:38: apps] Error 2
make[1]: Leaving directory '/home/mryel/crowsnest/bin/ustreamer'
make: *** [Makefile:28: all] Error 2
make: *** [Makefile:19: install] Error 2
```

mdevaev, 2024-03-02:
How looks that message about missing `atomic_ullong`? I thought it should be available in any case because of standard

mryel00, 2024-03-02:
This is the whole compiler output. I just looked through the different commits you made and came across those two changes I mentioned in the first message.
After changing `atomic_ullong` to `atomic_ulong` everything compiled again.

Adding `LDFLAGS=-latomic` seems to work too, without changing `atomic_ullong`.

mdevaev, 2024-03-02:
Aha. I've commited it to the master branch. Please test it.

mryel00, 2024-03-02:
It compiles again. Thank 😄

closed 2024-03-02 by mryel00

mdevaev, 2024-03-03:
:ok_hand: