pikvm/ustreamer #264: error: incompatible integer to pointer conversion passing 'int' to parameter of type 'const char *' [-Wint-conversion]

yurivict, 2024-03-04:
Build fails:
```
In file included from libs/base64.c:30:
libs/tools.h:192:19: error: incompatible integer to pointer conversion passing 'int' to parameter of type 'const char *' [-Wint-conversion]
        return us_strdup(strerror_r(error, buf, max_len));
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libs/tools.h:81:36: note: passing argument to parameter 'str' here
INLINE char *us_strdup(const char *str) {
                                   ^
1 error generated.
gmake[2]: *** [Makefile:125: build/libs/base64.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from libs/frametext.c:23:
In file included from libs/frametext.h:27:
In file included from libs/frame.h:26:
libs/tools.h:192:19: error: incompatible integer to pointer conversion passing 'int' to parameter of type 'const char *' [-Wint-conversion]
        return us_strdup(strerror_r(error, buf, max_len));
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libs/tools.h:81:36: note: passing argument to parameter 'str' here
INLINE char *us_strdup(const char *str) {
                                   ^
1 error generated.
gmake[2]: *** [Makefile:125: build/libs/frametext.o] Error 1
In file included from libs/device.c:23:
In file included from libs/device.h:28:
In file included from libs/frame.h:26:
libs/tools.h:192:19: error: incompatible integer to pointer conversion passing 'int' to parameter of type 'const char *' [-Wint-conversion]
        return us_strdup(strerror_r(error, buf, max_len));
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libs/tools.h:81:36: note: passing argument to parameter 'str' here
INLINE char *us_strdup(const char *str) {
                                   ^
1 error generated.
```

Version: 5.58
clang-16
FreeBSD 14.0

mdevaev, 2024-03-05:
It should be fixed. Please check the master branch.

mdevaev, 2024-03-05:
Updated again.

mdevaev, 2024-03-09:
Ping?

mdevaev, 2024-03-11:
:ok_hand: