pikvm/ustreamer #268: CAP: Can't release HW buffer=1 (skipped frame): Cannot allocate memory

breakingflower, 2024-03-26:
I'm getting the following on an RPI zero 2w and camera v3 noir: 
Device: 
- RPI Zero 2W
- Camera: v3 module NoIR
- os: raspbian (debian) 12 

Steps to reproduce: 
- fresh install latest raspbian image 
Do upgrades and install modules 
```
sudo apt update && sudo apt -y upgrade
sudo apt install build-essential libevent-dev libjpeg-dev libbsd-dev git
git clone --depth=1 https://github.com/pikvm/ustreamer
cd ustreamer
make
sudo apt -y install libcamera-tools libcamera-v4l2
sudo modprobe bcm2835-v4l2
libcamerify ./ustreamer --host :: --encoder=m2m-image
```
Output: 
```
pi@webcampi:~/ustreamer $ libcamerify ./ustreamer --host :: --encoder=m2m-image --debug
-- INFO  [220.678          ] -- Starting PiKVM uStreamer 6.7 ...
-- DEBUG [220.679          ] -- Installing SIGINT handler ...
-- DEBUG [220.679          ] -- Installing SIGTERM handler ...
-- DEBUG [220.679          ] -- Ignoring SIGPIPE ...
-- DEBUG [220.679          ] -- HTTP: Binding HTTP to [::]:8080 ...
-- INFO  [220.679          ] -- HTTP: Listening HTTP on [::]:8080
-- INFO  [220.680          ] -- Using V4L2 device: /dev/video0
-- INFO  [220.680          ] -- Using desired FPS: 0
-- INFO  [220.680          ] -- HTTP: Starting eventloop ...
-- DEBUG [220.680          ] -- CAP: Opening capture device ...
[0:03:40.681904693] [1439] ERROR IPAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
[0:03:40.682034484] [1439] ERROR IPAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
[0:03:40.682168024] [1439]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
-- DEBUG [220.696          ] -- HTTP: Repeating exposed ...
[0:03:40.837625987] [1441]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:03:40.843194123] [1441]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
-- DEBUG [220.844          ] -- CAP: Capture device fd=8 opened
-- DEBUG [220.844          ] -- CAP: Querying device capabilities ...
-- INFO  [220.844          ] -- CAP: Using capture type: single-planar
-- INFO  [220.844          ] -- CAP: Using input channel: 0
-- DEBUG [220.845          ] -- CAP: Using TV standard: DEFAULT
-- DEBUG [220.845          ] -- CAP: Probing device format=YUYV, stride=1280, resolution=640x480 ...
[0:03:40.845366450] [1439]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:03:40.846031133] [1441]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
-- INFO  [220.848          ] -- CAP: Using resolution: 640x480
-- INFO  [220.849          ] -- CAP: Using format: YUYV
-- DEBUG [220.849          ] -- CAP: Querying HW FPS ...
-- INFO  [220.849          ] -- CAP: Querying HW FPS changing is not supported
-- INFO  [220.849          ] -- CAP: Using IO method: MMAP
-- DEBUG [220.849          ] -- CAP: Requesting 5 device buffers for MMAP ...
[0:03:40.849082829] [1439]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:03:40.849748866] [1441]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
-- INFO  [220.855          ] -- CAP: Requested 5 device buffers, got 5
-- DEBUG [220.856          ] -- CAP: Allocating device buffers ...
-- DEBUG [220.856          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=0 ...
-- DEBUG [220.856          ] -- CAP: Mapping device buffer=0 ...
-- DEBUG [220.856          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=1 ...
-- DEBUG [220.856          ] -- CAP: Mapping device buffer=1 ...
-- DEBUG [220.856          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=2 ...
-- DEBUG [220.856          ] -- CAP: Mapping device buffer=2 ...
-- DEBUG [220.856          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=3 ...
-- DEBUG [220.856          ] -- CAP: Mapping device buffer=3 ...
-- DEBUG [220.856          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=4 ...
-- DEBUG [220.856          ] -- CAP: Mapping device buffer=4 ...
-- DEBUG [220.856          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=0 ...
-- DEBUG [220.856          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=1 ...
-- DEBUG [220.857          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=2 ...
-- DEBUG [220.857          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=3 ...
[... 214 lines cut ...]
-- DEBUG [227.888          ] -- CAP: Grabbing hw buffer ...
[0:03:47.888327480] [1439] ERROR V4L2Compat v4l2_camera_proxy.cpp:680 Failed to clear eventfd POLLIN
[0:03:47.888508729] [1439] ERROR V4L2Compat v4l2_camera_proxy.cpp:680 Failed to clear eventfd POLLIN
[0:03:47.888609562] [1439] ERROR Request request.cpp:472 FrameBuffer already set for stream
[0:03:47.888666228] [1439] ERROR V4L2Compat v4l2_camera.cpp:259 Can't set buffer for request
-- ERROR [227.889          ] -- CAP: Can't release HW buffer=1 (skipped frame): Cannot allocate memory
-- INFO  [227.960          ] -- Destroying workers pool JPEG ...
-- DEBUG [227.961          ] -- Bye-bye (worker jw-0)
-- DEBUG [227.961          ] -- Bye-bye (worker jw-1)
-- DEBUG [227.961          ] -- Bye-bye (worker jw-2)
-- DEBUG [227.961          ] -- Bye-bye (worker jw-3)
-- DEBUG [227.962          ] -- CAP: Calling VIDIOC_STREAMOFF ...
-- DEBUG [227.975          ] -- CAP: Releasing HW buffers ...
-- INFO  [227.978          ] -- CAP: Capturing stopped
================================================================================
-- DEBUG [227.979          ] -- CAP: Opening capture device ...
-- DEBUG [227.980          ] -- CAP: Capture device fd=8 opened
-- DEBUG [227.980          ] -- CAP: Querying device capabilities ...
-- INFO  [227.981          ] -- CAP: Using capture type: single-planar
-- INFO  [227.981          ] -- CAP: Using input channel: 0
-- DEBUG [227.981          ] -- CAP: Using TV standard: DEFAULT
-- DEBUG [227.981          ] -- CAP: Probing device format=YUYV, stride=1280, resolution=640x480 ...
```

It seems there is something happening with releasing memory. When I open the stream page I notice the first seconds images are changing, then the module breaks & starts errorring and can't restore itself.

davhenriques, 2024-06-02:
I am have the same hardware and having the same issue. Have you found any solution yet?

mdevaev, 2024-07-25:
Try `sudo modprobe bcm2835-v4l2` and then start ustreamer via libcamerify.

bread-mountain4, 2025-03-05:
I'm having this problem as well with latest ustreamer from source + libcamerify on Raspbian Bookworm with a Raspberry Pi Camera module 3. Tried the modprobe command but I still get the same errors from (what looks like) libcamerify and ustreamer

```
$ libcamerify ./ustreamer --device=/dev/video0 --host=0.0.0.0 --port=5000 --encoder=m2m-image
-- INFO  [1763.416          ] -- Creating pool JPEG with 4 workers ...
-- INFO  [1763.418          ] -- Capturing ...
[0:29:23.437622644] [5678] ERROR V4L2Compat v4l2_camera_proxy.cpp:705 Failed to clear eventfd POLLIN
[0:29:23.437778163] [5678] ERROR V4L2Compat v4l2_camera_proxy.cpp:705 Failed to clear eventfd POLLIN
[0:29:23.437840037] [5678] ERROR Request request.cpp:480 FrameBuffer already set for stream
[0:29:23.437881547] [5678] ERROR V4L2Compat v4l2_camera.cpp:263 Can't set buffer for request
-- ERROR [1763.438          ] -- CAP: Can't release HW buffer=1 (skipped frame): Cannot allocate memory
-- INFO  [1763.518          ] -- Destroying workers pool JPEG ...
-- INFO  [1763.535          ] -- CAP: Capturing stopped
```

mdevaev, 2025-03-09:
I think you need to ask about libcamerify here: https://github.com/raspberrypi/libcamera

uStreamer works with pure V4L2 and doesn't do anything weird with the kernel API, so if something doesn't work, then the problem is most likely on the libcamerify side. I'm open to suggestions if the bug is on my side, but the logs show that it doesn't look like it.