pikvm/ustreamer #271: Logitech C270 "Can't set input channel" on Ubuntu 22.04

jziolkowski, 2024-04-11:
Pretty much title. I've tried multiple channels, nothing works.

`Bus 001 Device 003: ID 046d:0825 Logitech, Inc. Webcam C270`

Output from v4l2-ctl:

```
Driver Info:
        Driver name      : uvcvideo
        Card type        : UVC Camera (046d:0825)
        Bus info         : usb-0000:00:1d.7-6
        Driver version   : 5.15.143
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : uvcvideo
        Model            : UVC Camera (046d:0825)
        Serial           : 4EA35A20
        Bus info         : usb-0000:00:1d.7-6
        Media version    : 5.15.143
        Hardware revision: 0x00000012 (18)
        Driver version   : 5.15.143
Interface Info:
        ID               : 0x03000002
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : UVC Camera (046d:0825)
        Function         : V4L2 I/O
        Flags            : default
        Pad 0x01000007   : 0: Sink
          Link 0x02000019: from remote pad 0x100000a of entity 'Extension 4' (Video Pixel Formatter): Data, Enabled, Immutable
Priority: 2
Video input : 0 (Input 1: ok)
Format Video Capture:
        Width/Height      : 1280/720
        Pixel Format      : 'MJPG' (Motion-JPEG)
        Field             : None
        Bytes per Line    : 0
        Size Image        : 816000
        Colorspace        : sRGB
        Transfer Function : Rec. 709
        YCbCr/HSV Encoding: ITU-R 601
        Quantization      : Default (maps to Full Range)
        Flags             :
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 1280, Height 720
        Default     : Left 0, Top 0, Width 1280, Height 720
        Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 1280, Height 720, Flags:
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1280, Height 720, Flags:
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Read buffers     : 0

User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=32 value=32
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=32 value=32
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=64 value=131
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2 (60 Hz)
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=0 max=10000 step=10 default=4000 value=2000 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=24 value=24
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
                                1: Manual Mode
                                3: Aperture Priority Mode
         exposure_time_absolute 0x009a0902 (int)    : min=1 max=10000 step=1 default=166 value=667 flags=inactive
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=1
                      led1_mode 0x0a046d05 (menu)   : min=0 max=3 default=3 value=3 (Auto)
                                0: Off
                                1: On
                                2: Blink
                                3: Auto
                 led1_frequency 0x0a046d06 (int)    : min=0 max=131 step=1 default=0 value=0
```

Camera is tested and known to work (both Windows and Linux)

Any pointers how I can at least try and start to debug that problem?

mdevaev, 2024-04-12:
Please try `ustreamer -m jpeg` or `mjpeg`

jziolkowski, 2024-04-12:
Ok, so I'm confused. I've swapped cable to a different USB port (secondary hub) and it works like a charm. Then I started poking around and noticed that this mobo is somewhat dated, and has two hubs, one of which is USB 1.1. Which is weird, because C270 should work in either, being a 1.1 device itself.

mdevaev, 2024-04-12:
Well, this is hardware problem. ustreamer does not use any hacks or custom drivers.

closed 2024-04-17 by mdevaev