pikvm/ustreamer #272: Does ustreamer support output_file.so for image capturing in a ring buffer like mjpg streamer?

gitterman, 2024-04-20:
I use mjpg streamer on a raspberry 3 as an offline surveillance solution.
For that, I capture images in a ring buffer on a RAM drive and only store these away, when a PIR sensor detects an event.
Thus, I have pre and post trigger images and only use my non-volatile drive when necessary.
This saves a lot of data space and only keeps events.
The saving of single images in a ring buffer is the most important feature for me, but in the comparison between ustreamer and mjpg streamer I do find nothing about file storage with output_file.so though.

mdevaev, 2024-04-25:
Hello. Ustreamer doesn't have such feature. However, we have a mechanism so that you can get up-to-date pictures of their shared memory. There is python binding. Will this suit you?

gitterman, 2024-04-25:
@mdevaev
The ring buffer feature is built in into mjpg_streamers output_file.so, and it can be very easily configured with Options 
 -f $PICfolder --size $RingBufferSize --exceed $RingbufferExceedPics 
It does even accept a post processing command.
This way it is very easy to capture pre-trigger events without any extra coding.
This is much easier and more space efficient than storing video snippets of a given length or size.
The Raspberry can easily copy 100 jpg pictures of ~80kB size from the RAM disk to non-volatile storage while capturing images in the ring butter continues in the background.
This way I do not lose any event and can process event tringgered data in the foreground. 
As ustreamer doesn't support output_file.so, it is no substitute for mjpg_streamer, at least not for me.

mdevaev, 2024-04-25:
Well, there is no such feature, but you can use memory sink, as I mentioned.

closed 2024-04-25 by mdevaev