pikvm/ustreamer #274: CAP:Device select() timeout

KawSplice, 2024-05-26:
================================================================================
-- INFO  [1340.939          ] -- Using V4L2 device: /dev/video0
-- INFO  [1340.939          ] -- CAP: Using capture type: single-planar
-- INFO  [1340.939          ] -- CAP: Using input channel: 0
-- INFO  [1340.939          ] -- CAP: Using resolution: 1980x1080
-- INFO  [1340.939          ] -- CAP: Using format: YUYV
-- INFO  [1340.939          ] -- CAP: Querying HW FPS changing is not supported
-- INFO  [1340.939          ] -- CAP: Using IO method: MMAP
-- INFO  [1340.958          ] -- CAP: Requested 5 device buffers, got 5
-- INFO  [1340.959          ] -- CAP: Capturing started
-- INFO  [1340.959          ] -- Using JPEG quality: 80%
-- INFO  [1340.959          ] -- Creating pool JPEG with 4 workers ...
-- INFO  [1340.961          ] -- Capturing ...
-- ERROR [1341.962          ] -- CAP: Device select() timeout
-- INFO  [1342.062          ] -- Destroying workers pool JPEG ...
-- INFO  [1342.069          ] -- CAP: Capturing stopped

mdevaev, 2024-05-26:
If you want to get some kind of support, you could describe the essence of the problem in more detail.

James4Ever0, 2024-09-08:
#278

James4Ever0, 2024-09-08:
@KawSplice 

Solution: https://github.com/jacksonliam/mjpg-streamer/issues/182#issuecomment-2336583229