pikvm/ustreamer #278: CAP: Device select() timeout

Foxius, 2024-06-14:
```
-- INFO  [4318.843          ] -- Using V4L2 device: /dev/v4l/by-id/usb-SHENZHEN_AONI_ELECTRONIC_CO._LTD_Full_HD_webcam_AN20200521001-video-index0
-- INFO  [4318.843          ] -- CAP: Using capture type: single-planar
-- INFO  [4318.843          ] -- CAP: Using input channel: 0
-- INFO  [4318.909          ] -- CAP: Using resolution: 1280x720
-- INFO  [4318.909          ] -- CAP: Using format: MJPEG
-- INFO  [4318.971          ] -- CAP: Using HW FPS: 30
-- ERROR [4318.971          ] -- CAP: Device doesn't support setting of HW encoding quality parameters
-- INFO  [4318.971          ] -- CAP: Using IO method: MMAP
-- INFO  [4318.973          ] -- CAP: Requested 3 device buffers, got 3
-- INFO  [4318.993          ] -- CAP: Capturing started
-- INFO  [4318.993          ] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [4318.993          ] -- Using JPEG quality: encoder default
-- INFO  [4318.993          ] -- Creating pool JPEG with 1 workers ...
-- INFO  [4318.993          ] -- Capturing ...
^C-- INFO  [4319.055          ] -- ===== Stopping by SIGINT =====
-- INFO  [4319.055          ] -- HTTP: Eventloop stopped
-- ERROR [4319.995          ] -- CAP: Device select() timeout
-- INFO  [4320.000          ] -- Destroying workers pool JPEG ...
-- INFO  [4320.064          ] -- CAP: Capturing stopped
-- INFO  [4320.065          ] -- Bye-bye
```
```
saikonohack@raspberry:~/ustreamer $ v4l2-ctl -d /dev/v4l/by-id/usb-SHENZHEN_AONI_ELECTRONIC_CO._LTD_Full_HD_webcam_AN20200521001-video-index0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 1280x960
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                        Interval: Discrete 0.333s (3.000 fps)
                Size: Discrete 1280x960
                        Interval: Discrete 0.200s (5.000 fps)
                        Interval: Discrete 0.333s (3.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.200s (5.000 fps)
```
`./ustreamer --device=/dev/v4l/by-id/usb-SHENZHEN_AONI_ELECTRONIC_CO._LTD_Full_HD_webcam_AN20200521001-video-index0 --host=0.0.0.0 --port=8080 --resolution 1280x720 --desired-fps=30 --format=MJPEG --drop-same-frames=30`

mdevaev, 2024-06-15:
Is it working with other software like mjpg-streamer?

Foxius, 2024-06-15:
I honestly don't know how to use mjpeg-streamer. But I for example wrote a script in python and translated it to browser. it works but not for klipper. (but for example in telegram addon the same link works)

mdevaev, 2024-06-15:
Wait a moment. Before you're hitting Ctrl+C, do you have a video stream?

Foxius, 2024-06-15:
no

mdevaev, 2024-06-15:
Did you try --debug?

mdevaev, 2024-07-07:
No feedback, closing.

closed 2024-07-07 by mdevaev

James4Ever0, 2024-09-08:
Here is everything after `--debug` appended to commandline:

```
-- DEBUG [8491.973          ] -- CAP: Opening capture device ...
-- DEBUG [8491.973          ] -- CAP: Capture device fd=8 opened
-- INFO  [8491.974          ] -- Using V4L2 device: /dev/kvmd-video
-- DEBUG [8491.974          ] -- CAP: Querying device capabilities ...
-- INFO  [8491.974          ] -- CAP: Using capture type: single-planar
-- INFO  [8491.974          ] -- CAP: Using input channel: 0
-- DEBUG [8491.974          ] -- CAP: Using TV standard: DEFAULT
-- DEBUG [8491.974          ] -- CAP: Probing device format=MJPEG, stride=3840, resolution=1920x1080 ...
-- INFO  [8491.976          ] -- CAP: Using resolution: 1920x1080
-- INFO  [8491.977          ] -- CAP: Using format: MJPEG
-- DEBUG [8491.977          ] -- CAP: Querying HW FPS ...
-- INFO  [8491.979          ] -- CAP: Using HW FPS: 30
-- ERROR [8491.979          ] -- CAP: Device doesn't support setting of HW encoding quality parameters
-- INFO  [8491.979          ] -- CAP: Using IO method: MMAP
-- DEBUG [8491.979          ] -- CAP: Requesting 5 device buffers for MMAP ...
-- INFO  [8491.996          ] -- CAP: Requested 5 device buffers, got 5
-- DEBUG [8491.996          ] -- CAP: Allocating device buffers ...
-- DEBUG [8491.996          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=0 ...
-- DEBUG [8491.996          ] -- CAP: Mapping device buffer=0 ...
-- DEBUG [8491.997          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=1 ...
-- DEBUG [8491.997          ] -- CAP: Mapping device buffer=1 ...
-- DEBUG [8491.999          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=2 ...
-- DEBUG [8491.999          ] -- CAP: Mapping device buffer=2 ...
-- DEBUG [8492.000          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=3 ...
-- DEBUG [8492.001          ] -- CAP: Mapping device buffer=3 ...
-- DEBUG [8492.002          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=4 ...
-- DEBUG [8492.002          ] -- CAP: Mapping device buffer=4 ...
-- DEBUG [8492.003          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=0 ...
-- DEBUG [8492.003          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=1 ...
-- DEBUG [8492.003          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=2 ...
-- DEBUG [8492.003          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=3 ...
-- DEBUG [8492.003          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=4 ...
-- INFO  [8492.006          ] -- CAP: Capturing started
-- INFO  [8492.006          ] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [8492.006          ] -- Using JPEG quality: encoder default
-- INFO  [8492.007          ] -- Creating pool JPEG with 1 workers ...
-- DEBUG [8492.007          ] -- Hello! I am a worker jw-0 ^_^
-- DEBUG [8492.008          ] -- Worker jw-0 waiting for a new job ...
-- INFO  [8492.009          ] -- Capturing ...
-- DEBUG [8492.009          ] -- CAP: Calling select() on video device ...
-- DEBUG [8492.009          ] -- CAP: Device select() --> 1; has_read=1, has_error=0
-- DEBUG [8492.009          ] -- CAP: Grabbing hw buffer ...
-- DEBUG [8492.010          ] -- CAP: Dropped too small frame, assuming it was broken: buffer=0, bytesused=0
-- DEBUG [8492.010          ] -- CAP: Releasing HW buffer=0 (broken frame) ...
-- DEBUG [8492.010          ] -- CAP: Dropped too small frame, assuming it was broken: buffer=1, bytesused=0
-- DEBUG [8492.011          ] -- CAP: Releasing HW buffer=1 (broken frame) ...
-- DEBUG [8492.011          ] -- CAP: Dropped too small frame, assuming it was broken: buffer=2, bytesused=0
-- DEBUG [8492.011          ] -- CAP: Releasing HW buffer=2 (broken frame) ...
-- DEBUG [8492.011          ] -- CAP: Dropped too small frame, assuming it was broken: buffer=3, bytesused=0
-- DEBUG [8492.011          ] -- CAP: Releasing HW buffer=3 (broken frame) ...
-- DEBUG [8492.011          ] -- CAP: Dropped too small frame, assuming it was broken: buffer=4, bytesused=0
-- DEBUG [8492.011          ] -- CAP: Releasing HW buffer=4 (broken frame) ...
-- DEBUG [8492.011          ] -- CAP: Calling select() on video device ...
-- DEBUG [8492.593          ] -- HTTP: Repeating exposed ...
-- DEBUG [8493.012          ] -- CAP: Device select() --> 0; has_read=0, has_error=0
-- ERROR [8493.012          ] -- CAP: Device select() timeout
-- INFO  [8493.110          ] -- Destroying workers pool JPEG ...
-- DEBUG [8493.110          ] -- Bye-bye (worker jw-0)
-- DEBUG [8493.111          ] -- CAP: Calling VIDIOC_STREAMOFF ...
-- DEBUG [8493.112          ] -- CAP: Releasing HW buffers ...
-- INFO  [8493.122          ] -- CAP: Capturing stopped
```

James4Ever0, 2024-09-08:
@Foxius 

Solution: https://github.com/jacksonliam/mjpg-streamer/issues/182#issuecomment-2336583229