pikvm/ustreamer #279: UnixClientConnectorError

TanZQ, 2024-06-18:
I encountered some problems when I tried to access the pikvm device through the API and obtain screenshots. 

The snapshot request of the API received a 503 error, which was displayed in the kvmd log of pikvm as:
`Jun 18 08:46:40 pikvm kvmd[1703]: kvmd.apps.kvmd.api.streamer      ERROR --- Can't connect to streamer: UnixClientConnectorError: Cannot connect to unix socket /run/kvmd/ustreamer.sock ssl:default [No such file or directory]`

and the api access file is like:
`import requests
from PIL import Image
from io import BytesIO
import warnings
from urllib3.exceptions import InsecureRequestWarning

warnings.simplefilter('ignore', InsecureRequestWarning)

PIKVM_IP = "133.68.**.***"
API_ENDPOINT = f"https://{PIKVM_IP}/api/streamer/snapshot" 

params = {
    "save": "false",
    "load": "false",
    "allow_offline": "true",
    "ocr": "false",
    "preview": "false"
}

try:
    response = requests.get(API_ENDPOINT, params=params, verify=False, auth=('admin', 'admin'))
    response.raise_for_status()  # Raise an exception for HTTP errors
    if response.status_code == 200:
        image = Image.open(BytesIO(response.content))
        image.show()
        image.save("screenshot.jpg")
        print("Screenshot saved as screenshot.jpg")
    else:
        print(f"Failed to get screenshot. Status code: {response.status_code}, Response: {response.text}")
except requests.exceptions.RequestException as e:
    print(f"Request failed: {e}")`

I tried some methods but still cant solve this problem.

Looking forward for your help!

mdevaev, 2024-06-18:
Hello. To take a screenshot you need to start ustreamer first. So, you need to open /api/ws connection, keep it for few seconds and then take a screenshot via api.

TanZQ, 2024-06-19:
> Hello. To take a screenshot you need to start ustreamer first. So, you need to open websocket /api/ws connection, wait for few seconds and then take a screenshot via api.

Thank you very much for your prompt answer, it seems that I didn't read the API part of the PIKVM handbook carefully. My problem has been solved.

TanZQ, 2024-06-19:
> Hello. To take a screenshot you need to start ustreamer first. So, you need to open websocket /api/ws connection, wait for few seconds and then take a screenshot via api.
Sorry to bother you again, I would like to ask where can I find the keyword definitions for all websocket events (including mouse, keyboard, etc.)?
The handbook says "Each event type will be described in the corresponding paragraph for its component", but I can't find where I should look them up.
In the example, I know that the keywords for keyboard events are:
event = {"event_type": "key","event": {"key": key,"state": state}}
except that, I have not yet found what values ​​should be filled in the json block sent when performing other keyboard and mouse operations.
If possible, please remind me where to find them, thank you!

mdevaev, 2024-06-19:
They are here: https://developer.mozilla.org/en-US/docs/Web/API/KeyboardEvent/code/code_values

TanZQ, 2024-06-19:
> They are here: https://developer.mozilla.org/en-US/docs/Web/API/KeyboardEvent/code/code_values

Sorry, maybe I didn't express myself clearly.
For example, regarding mouse movement via websocket, I want to ask:
Is that` "event_type": "mousemove" `or `"event_type": "mouse_move"` or something else
Is that` "event": {"move": "x,y"}` or` "event": {"to": "x,y"}` or something else
How can i figure these formats out?

mdevaev, 2024-06-19:
Here you can find all HID API definitions: https://github.com/pikvm/kvmd/blob/master/kvmd/apps/kvmd/api/hid.py

The correct event: `ws.send('{"event_type": "key", "event": {"key": "Enter", "state": true}}')`

This is described in the example: https://docs.pikvm.org/api/#websocket-events

closed 2024-06-20 by mdevaev