pikvm/ustreamer #280: Use VGA Acquisition card will Image dislocation

sksorth, 2024-06-20:
My parameter configuration is：ustreamer --device=/dev/video9 --host=0.0.0.0 --port=8010 --allow-origin=\* -r 1920x1080 -f 60 -m MJPEG -c M2M-VIDEO --debug &

such as：

[image]
[image]

mdevaev, 2024-06-20:
Probably this is a driver problem.

sksorth, 2024-06-21:
But it will not appear on the HTMI collection card,If you use the frame rate of 30fps on VGA, it will occasionally appear several times, for example: 3 minutes 5 times.

sksorth, 2024-06-21:
And I use FFMPEG to collect, there will be no such situation.

mdevaev, 2024-06-21:
Try to remove option `-c M2M-VIDEO`

sksorth, 2024-06-25:
I changed a development board to solve, it may indeed be a driver's problem.

closed 2024-06-25 by sksorth