pikvm/ustreamer #281: ERRO: CAP: Device doesn't support setting of HW encoding quality parameters

Vittuu, 2024-07-12:
Hello, I have a problem with my Orange Pi Zero3 4GB and a camera called Studio P5. It works fine in YUYV mode, but I want to use the 1920x1080p resolution. However, I always get this error when trying to execute the following command:

orangepi@orangepi:~/ustreamer$ sudo ./ustreamer --device=/dev/video1 --host=0.0.0.0 --port=80 --resolution 1920x1080 -m mjpeg

```
[sudo] senha para orangepi: 
-- INFO  [2673.839          ] -- Starting PiKVM uStreamer 6.12 ...
-- INFO  [2673.840          ] -- HTTP: Listening HTTP on [0.0.0.0]:80
-- INFO  [2673.840          ] -- HTTP: Starting eventloop ...
-- INFO  [2673.977          ] -- Using V4L2 device: /dev/video1
-- INFO  [2673.977          ] -- CAP: Using capture type: single-planar
-- INFO  [2673.977          ] -- CAP: Using input channel: 0
-- INFO  [2673.978          ] -- CAP: Using resolution: 1920x1080
-- INFO  [2673.978          ] -- CAP: Using format: MJPEG
-- INFO  [2673.983          ] -- CAP: Using HW FPS: 0 -> 24 (coerced)
-- ERROR [2673.983          ] -- CAP: Device doesn't support setting of HW encoding quality parameters
-- INFO  [2673.983          ] -- CAP: Using IO method: MMAP
-- INFO  [2673.986          ] -- CAP: Requested 5 device buffers, got 5
-- INFO  [2676.615          ] -- CAP: Capturing started
-- INFO  [2676.615          ] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [2676.615          ] -- Using JPEG quality: encoder default
-- INFO  [2676.615          ] -- Creating pool JPEG with 1 workers ...
-- INFO  [2676.616          ] -- Capturing ...
-- INFO  [2688.030          ] -- HTTP: NEW client (now=1): [192.168.246.248]:61192, id=f64d95f9a6b087f3
^C-- INFO  [2745.143          ] -- ===== Stopping by SIGINT =====
-- INFO  [2745.143          ] -- HTTP: Eventloop stopped
-- INFO  [2745.237          ] -- Destroying workers pool JPEG ...
-- INFO  [2745.535          ] -- CAP: Capturing stopped
-- INFO  [2745.536          ] -- Bye-bye
```
this is my informations of de camera studio p5

`orangepi@orangepi:~/ustreamer$ v4l2-ctl --all -d /dev/video1
Driver Info:
	Driver name      : uvcvideo
	Card type        : Poly Studio P5 webcam
	Bus info         : usb-5200000.usb-1.1
	Driver version   : 6.1.31
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : Poly Studio P5 webcam
	Serial           : 212614797
	Bus info         : usb-5200000.usb-1.1
	Media version    : 6.1.31
	Hardware revision: 0x00000198 (408)
	Driver version   : 6.1.31
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : Poly Studio P5 webcam
	Function         : V4L2 I/O
	Flags            : default
	Pad 0x01000007   : 0: Sink
	  Link 0x02000019: from remote pad 0x100000a of entity 'Extension 2' (Video Pixel Formatter): Data, Enabled, Immutable
Priority: 2
Video input : 0 (Input 1: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'YUYV' (YUYV 4:2:2)
	Field             : None
	Bytes per Line    : 1280
	Size Image        : 460800
	Colorspace        : sRGB
	Transfer Function : Rec. 709
	YCbCr/HSV Encoding: ITU-R 601
	Quantization      : Default (maps to Limited Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 480
	Default     : Left 0, Top 0, Width 640, Height 480
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 640, Height 480, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 480, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0

User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=128
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=128 value=128
                            hue 0x00980903 (int)    : min=-128 max=128 step=1 default=0 value=0
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=1 max=255 step=1 default=128 value=128
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2 (60 Hz)
				0: Disabled
				1: 50 Hz
				2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2000 max=7500 step=10 default=4000 value=4000 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=128 value=128
         backlight_compensation 0x0098091c (int)    : min=0 max=4 step=1 default=1 value=1

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
				1: Manual Mode
				3: Aperture Priority Mode
         exposure_time_absolute 0x009a0902 (int)    : min=3 max=2047 step=1 default=3 value=3 flags=inactive
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=0
                   pan_absolute 0x009a0908 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                 focus_absolute 0x009a090a (int)    : min=0 max=1023 step=1 default=0 value=0 flags=inactive
     focus_automatic_continuous 0x009a090c (bool)   : default=1 value=1
                  zoom_absolute 0x009a090d (int)    : min=10 max=40 step=1 default=10 value=10
                        privacy 0x009a0910 (bool)   : default=0 value=0
`
But if i use de yuvy work correct:

`orangepi@orangepi:~/Área de Trabalho/Ustreamer/ustreamer-6.12$ sudo ./ustreamer --device=/dev/video1 --host=0.0.0.0 --port=80
-- INFO  [11922.827          ] -- Starting PiKVM uStreamer 6.12 ...
-- INFO  [11922.827          ] -- HTTP: Listening HTTP on [0.0.0.0]:80
-- INFO  [11922.828          ] -- HTTP: Starting eventloop ...
-- INFO  [11922.961          ] -- Using V4L2 device: /dev/video1
-- INFO  [11922.961          ] -- CAP: Using capture type: single-planar
-- INFO  [11922.961          ] -- CAP: Using input channel: 0
-- INFO  [11922.962          ] -- CAP: Using resolution: 640x480
-- INFO  [11922.962          ] -- CAP: Using format: YUYV
-- INFO  [11922.967          ] -- CAP: Using HW FPS: 0 -> 30 (coerced)
-- INFO  [11922.967          ] -- CAP: Using IO method: MMAP
-- INFO  [11922.968          ] -- CAP: Requested 5 device buffers, got 5
-- INFO  [11923.604          ] -- CAP: Capturing started
-- INFO  [11923.604          ] -- Using JPEG quality: 80%
-- INFO  [11923.604          ] -- Creating pool JPEG with 4 workers ...
-- INFO  [11923.605          ] -- Capturing ...
-- INFO  [11930.824          ] -- HTTP: NEW client (now=1): [127.0.0.1]:45716, id=2e3097dd579160df
^C-- INFO  [12047.393          ] -- ===== Stopping by SIGINT =====
-- INFO  [12047.394          ] -- HTTP: Eventloop stopped
-- INFO  [12047.485          ] -- Destroying workers pool JPEG ...
^B-- INFO  [12047.752          ] -- CAP: Capturing stopped
-- INFO  [12047.753          ] -- Bye-bye`

this the formats my camera suports:

orangepi@orangepi:~/ustreamer$ v4l2-ctl -d /dev/video1 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 2560x1440
			Interval: Discrete 0.500s (2.000 fps)
		Size: Discrete 480x270
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 720x480
			Interval: Discrete 0.067s (15.000 fps)
	[1]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1024x576
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1600x896
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.042s (24.000 fps)
		Size: Discrete 2560x1440
			Interval: Discrete 0.042s (24.000 fps)
		Size: Discrete 480x270
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 720x480
			Interval: Discrete 0.033s (30.000 fps)
`

mdevaev, 2024-07-12:
This is not a fatal error, but only a message that your camera does not support hardware FPS settings.

Vittuu, 2024-07-12:
But isn't there a way for me to achieve this video at an acceptable FPS? It only delivers an image at almost -1 FPS.

mdevaev, 2024-07-12:
Low FPS is an unrelated issue with this message. Perhaps your camera is just giving out frames slowly, or there is some kind of network problem.

Vittuu, 2024-07-13:
I have already tested the internet; I just don't understand why in MJPEG it is showing this error along with the low refresh rate. It doesn't allow me to increase the number of workers either, while in YUVY this doesn't happen, but I can't get good quality in YUVY, Is there any command for me to switch from HW to another type of render?

mdevaev, 2024-07-14:
MJPEG is generated on the camera side by hardware. In this case, ustreamer simply transmits the data from it to you. The number of workers cannot be increased, because the workers simply have nothing to do: one worker transfers bytes from one buffer to another.

--format=YUYV can be used along with --encoder=CPU, and with this you can change workers and quality level.

The error you see about HW quality does not concern FPS (it was a typo). This means that for MJPEG you cannot change the compression quality because the hardware does not support it. For YUYV, this message is not displayed because the compression will be handled by the processor, where the quality can be adjusted.

You can try MJPEG and check http `/status` handle to see how fast the image is being captured from the camera.

Vittuu, 2024-07-14:
here is a reciver in /status:
`{"ok": true, "result": { "instance_id": "", "encoder": {"type": "HW", "quality": 0}, "source": {"resolution": {"width": 1920, "height": 1080}, "online": true, "desired_fps": 0, "captured_fps": 2}, "stream": {"queued_fps": 2, "clients": 5, "clients_stat": {"f48cc4ccda100f2c": {"fps": 2, "extra_headers": false, "advance_headers": false, "dual_final_frames": false, "zero_data": false, "key": "0"}, "2f3c78c9d7c9d9f3": {"fps": 2, "extra_headers": false, "advance_headers": false, "dual_final_frames": false, "zero_data": false, "key": "0"}, "67203fdeb71a8e8": {"fps": 2, "extra_headers": false, "advance_headers": false, "dual_final_frames": false, "zero_data": false, "key": "0"}, "751a50d703148bb7": {"fps": 2, "extra_headers": false, "advance_headers": false, "dual_final_frames": false, "zero_data": false, "key": "0"}, "4c1ffee0e915b2fe": {"fps": 2, "extra_headers": false, "advance_headers": false, "dual_final_frames": false, "zero_data": false, "key": "0"}}}}}`

what i can do with this?

mdevaev, 2024-07-15:
`"captured_fps": 2`

So your camera generates the slow video stream for some reason. Did you try any other software? Because now this is looking as a device-side problem.

Vittuu, 2024-07-15:
No, this was the first one I managed to get working on Linux. Is there any you recommend?

mdevaev, 2024-07-15:
You can try mjpg-streamer or VLC

mdevaev, 2024-07-23:
Answared I guess.

closed 2024-07-23 by mdevaev