pikvm/ustreamer #284: Repeated  `CAP: Device select() timeout`

jamincollins, 2024-08-06:
I have an Arducam 5MP Camera and can't seem to get it streaming reliably under ustreamer.

Device details:

```
$ v4l2-ctl --list-devices
Arducam 5MP Camera Module: Ardu (usb-0000:c2:00.3-1.2):
	/dev/video0
	/dev/video1
	/dev/media0
```

```
$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1024x768
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x960
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1600x1200
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 2048x1536
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 2592x1944
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.050s (20.000 fps)
		Size: Discrete 1024x768
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 1280x960
			Interval: Discrete 0.133s (7.500 fps)
		Size: Discrete 1600x1200
			Interval: Discrete 0.133s (7.500 fps)
		Size: Discrete 2048x1536
			Interval: Discrete 0.500s (2.000 fps)
		Size: Discrete 2592x1944
			Interval: Discrete 0.500s (2.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.200s (5.000 fps)
```

I can get a consistent stream from it via `vlc` with the following command:
```
$ vlc v4l2:///dev/video0:chroma=MJPEG
```

When trying stream through `ustreamer`:
```
$ ./ustreamer
-- INFO  [543179.118          ] -- Starting PiKVM uStreamer 6.12 ...
[... 441 lines cut ...]
-- DEBUG [543243.544          ] -- CAP: Device select() --> 1; has_read=1, has_error=0
-- DEBUG [543243.544          ] -- CAP: Grabbing hw buffer ...
-- DEBUG [543243.544          ] -- CAP: Grabbed HW buffer=1: bytesused=88595, grab_ts=543243.492, latency=0.052, skipped=0
-- DEBUG [543243.544          ] -- CAP: Calling select() on video device ...
-- VERB  [543243.544          ] -- JPEG: Passed encoding because nobody is watching
-- DEBUG [543243.549          ] -- CAP: Releasing HW buffer=1 ...
-- DEBUG [543243.549          ] -- CAP: HW buffer=1 released
-- DEBUG [543243.746          ] -- HTTP: Repeating exposed ...
-- DEBUG [543243.992          ] -- CAP: Device select() --> 1; has_read=1, has_error=0
-- DEBUG [543243.992          ] -- CAP: Grabbing hw buffer ...
-- DEBUG [543243.992          ] -- CAP: Grabbed HW buffer=1: bytesused=88382, grab_ts=543243.912, latency=0.080, skipped=0
-- DEBUG [543243.992          ] -- CAP: Calling select() on video device ...
-- VERB  [543243.992          ] -- JPEG: Passed encoding because nobody is watching
-- DEBUG [543243.998          ] -- CAP: Releasing HW buffer=1 ...
-- DEBUG [543243.998          ] -- CAP: HW buffer=1 released
-- DEBUG [543244.756          ] -- HTTP: Repeating exposed ...
-- DEBUG [543244.994          ] -- CAP: Device select() --> 0; has_read=0, has_error=0
-- ERROR [543244.994          ] -- CAP: Device select() timeout
-- INFO  [543245.094          ] -- Destroying workers pool JPEG ...
-- DEBUG [543245.094          ] -- Bye-bye (worker jw-0)
-- DEBUG [543245.094          ] -- CAP: Calling VIDIOC_STREAMOFF ...
-- DEBUG [543245.120          ] -- CAP: Releasing HW buffers ...
-- INFO  [543245.120          ] -- CAP: Capturing stopped
================================================================================
```

mdevaev, 2024-08-07:
Is it showing something at least between timeouts?

jamincollins, 2024-08-07:
Yes a few frames then the no signal notice. Then this cycle repeats. Also worth noting that the mjpeg format works best. The other seems almost like still frames or very low frame rate even though the device details says it should support 30 fps at 640x480.

mdevaev, 2024-08-07:
There may be something wrong with the driver. Try `--device-timeout=10`.

jamincollins, 2024-08-07:
That does not appear to resolve the `CAP: Device select() timeout`.  

Here is a capture from the same camera using VLC, as you can see it has no trouble staying connected and getting decent frame rate for the resolution.  This is scaled down from the original 1600x1200 capture.

https://github.com/user-attachments/assets/b1f35fbd-005d-40f2-93e0-9532ec064485

mdevaev, 2024-08-07:
Did you use device timeout with `--m MJPEG`?

jamincollins, 2024-08-07:
Yes. Both the `ustreamer` and `vlc` testing has been done with the same _format_ option of MJPEG.

mdevaev, 2024-08-08:
I can't say now what's wrong with it. Could you try [mjpg-streamer](https://github.com/jacksonliam/mjpg-streamer)? uStreamer doesn't do anything weird with camera, but maybe there is some difference.

James4Ever0, 2024-09-08:
#278 #274

James4Ever0, 2024-09-08:
https://github.com/jacksonliam/mjpg-streamer/issues/182#issuecomment-2336575203

James4Ever0, 2024-09-08:
@jamincollins 

Solution: https://github.com/jacksonliam/mjpg-streamer/issues/182#issuecomment-2336583229

mdevaev, 2024-09-08:
@jamincollins Please check if this is working.

Mactarvish, 2024-10-30:
Same error with log:
```
-- INFO  [2546.600          ] -- Starting PiKVM uStreamer 6.16 ...
-- DEBUG [2546.600          ] -- Installing SIGINT handler ...
-- DEBUG [2546.600          ] -- Installing SIGTERM handler ...
-- DEBUG [2546.600          ] -- Ignoring SIGPIPE ...
-- DEBUG [2546.600          ] -- HTTP: Binding HTTP to [0.0.0.0]:8080 ...
-- INFO  [2546.600          ] -- HTTP: Listening HTTP on [0.0.0.0]:8080
-- DEBUG [2546.600          ] -- CAP: Opening capture device ...
-- INFO  [2546.600          ] -- HTTP: Starting eventloop ...
-- DEBUG [2546.615          ] -- HTTP: Repeating exposed ...
-- DEBUG [2546.701          ] -- CAP: Capture device fd=8 opened
-- INFO  [2546.701          ] -- Using V4L2 device: /dev/video2
-- DEBUG [2546.701          ] -- CAP: Querying device capabilities ...
-- INFO  [2546.701          ] -- CAP: Using capture type: single-planar
-- INFO  [2546.701          ] -- CAP: Using input channel: 0
-- DEBUG [2546.701          ] -- CAP: Using TV standard: DEFAULT
-- DEBUG [2546.701          ] -- CAP: Probing device format=YUYV, stride=1280, resolution=640x480 ...
-- INFO  [2546.725          ] -- CAP: Using resolution: 640x480
-- INFO  [2546.725          ] -- CAP: Using format: YUYV
-- DEBUG [2546.725          ] -- CAP: Querying HW FPS ...
-- INFO  [2546.749          ] -- CAP: Using HW FPS: 0 -> 30 (coerced)
-- INFO  [2546.750          ] -- CAP: Using IO method: MMAP
-- DEBUG [2546.750          ] -- CAP: Requesting 5 device buffers for MMAP ...
-- INFO  [2546.750          ] -- CAP: Requested 5 device buffers, got 5
-- DEBUG [2546.750          ] -- CAP: Allocating device buffers ...
-- DEBUG [2546.750          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=0 ...
-- DEBUG [2546.750          ] -- CAP: Mapping device buffer=0 ...
-- DEBUG [2546.750          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=1 ...
-- DEBUG [2546.750          ] -- CAP: Mapping device buffer=1 ...
-- DEBUG [2546.750          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=2 ...
-- DEBUG [2546.750          ] -- CAP: Mapping device buffer=2 ...
-- DEBUG [2546.750          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=3 ...
-- DEBUG [2546.750          ] -- CAP: Mapping device buffer=3 ...
-- DEBUG [2546.750          ] -- CAP: Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=4 ...
-- DEBUG [2546.750          ] -- CAP: Mapping device buffer=4 ...
-- DEBUG [2546.750          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=0 ...
-- DEBUG [2546.750          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=1 ...
-- DEBUG [2546.750          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=2 ...
-- DEBUG [2546.750          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=3 ...
-- DEBUG [2546.750          ] -- CAP: Calling us_xioctl(VIDIOC_QBUF) for buffer=4 ...
-- INFO  [2546.764          ] -- CAP: Capturing started
-- INFO  [2546.764          ] -- Using JPEG quality: 80%
-- INFO  [2546.764          ] -- Creating pool JPEG with 4 workers ...
-- DEBUG [2546.765          ] -- Hello! I am a worker jw-0 ^_^
-- DEBUG [2546.765          ] -- Worker jw-0 waiting for a new job ...
-- DEBUG [2546.765          ] -- Hello! I am a worker jw-1 ^_^
-- DEBUG [2546.765          ] -- Worker jw-1 waiting for a new job ...
-- INFO  [2546.765          ] -- Capturing ...
-- DEBUG [2546.765          ] -- CAP: Calling select() on video device ...
-- DEBUG [2546.765          ] -- Hello! I am a worker jw-2 ^_^
-- DEBUG [2546.765          ] -- Worker jw-2 waiting for a new job ...
-- DEBUG [2546.765          ] -- Hello! I am a worker jw-3 ^_^
-- DEBUG [2546.765          ] -- Worker jw-3 waiting for a new job ...
-- DEBUG [2547.623          ] -- HTTP: Repeating exposed ...
-- DEBUG [2547.766          ] -- CAP: Device select() --> 0; has_read=0, has_error=0
-- ERROR [2547.766          ] -- CAP: Device select() timeout
-- INFO  [2547.866          ] -- Destroying workers pool JPEG ...
-- DEBUG [2547.867          ] -- Bye-bye (worker jw-0)
-- DEBUG [2547.867          ] -- Bye-bye (worker jw-1)
-- DEBUG [2547.867          ] -- Bye-bye (worker jw-2)
-- DEBUG [2547.867          ] -- Bye-bye (worker jw-3)
-- DEBUG [2547.867          ] -- CAP: Calling VIDIOC_STREAMOFF ...
-- DEBUG [2547.974          ] -- CAP: Releasing HW buffers ...
-- INFO  [2547.974          ] -- CAP: Capturing stopped
================================================================================
-- DEBUG [2547.974          ] -- CAP: Opening capture device ...
-- DEBUG [2547.974          ] -- CAP: Capture device fd=8 opened
-- INFO  [2547.974          ] -- Using V4L2 device: /dev/video2
-- DEBUG [2547.974          ] -- CAP: Querying device capabilities ...
[... 202 lines cut ...]
-- DEBUG [2551.926          ] -- Worker jw-0 waiting for a new job ...
-- DEBUG [2551.926          ] -- Hello! I am a worker jw-2 ^_^
-- DEBUG [2551.926          ] -- Worker jw-2 waiting for a new job ...
-- INFO  [2551.926          ] -- Capturing ...
-- DEBUG [2551.926          ] -- CAP: Calling select() on video device ...
-- DEBUG [2551.926          ] -- Hello! I am a worker jw-3 ^_^
-- DEBUG [2551.926          ] -- Worker jw-3 waiting for a new job ...
-- INFO  [2552.021          ] -- ===== Stopping by SIGINT =====
-- INFO  [2552.021          ] -- HTTP: Eventloop stopped
-- DEBUG [2552.926          ] -- CAP: Device select() --> 0; has_read=0, has_error=0
[... cut ...]
-- DEBUG [2552.927          ] -- Bye-bye (worker jw-2)
-- DEBUG [2552.928          ] -- Bye-bye (worker jw-3)
-- DEBUG [2552.928          ] -- CAP: Calling VIDIOC_STREAMOFF ...
-- DEBUG [2553.025          ] -- CAP: Releasing HW buffers ...
-- INFO  [2553.025          ] -- CAP: Capturing stopped
-- INFO  [2553.026          ] -- Bye-bye
```

mjpg-streamer works well.
The camera is usb2.0, while my laptop inplace camera works well.
https://github.com/jacksonliam/mjpg-streamer/issues/182#issuecomment-2336583229 doesn't work for me.

johnwc, 2024-12-06:
@jamincollins This is doing the same thing for us, looping with `CAP: Device select() timeout` We are using [this camera](https://www.amazon.com/gp/product/B09BR1RNSN/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1)