pikvm/ustreamer #286: Minor regression in 6.13: make fails when pkg-config isn't found

yurivict, 2024-08-16:
Build in the VM failed because pkg-config isn't present:
```
==>  Building for ustreamer-6.13
Makefile:21: *** recipe commences before first target.  Stop.
===> Compilation failed unexpectedly.
```

This is a minor problem, but I am still reporting it.

mdevaev, 2024-08-16:
It's needed for gpio, drm and janus plugin.

yurivict, 2024-08-16:
Yes, but gmake fails ungracegully when pkg-config is accidentally missing.

mdevaev, 2024-08-17:
Could you check the latest branch? It should be checked correctly.

yurivict, 2024-08-17:
The latest commit fails gracefully:
```
===>  Building for ustreamer-6.13
Makefile:21: *** "No pkg-config found in /usr/local/libexec/ccache:/usr/ports/multimedia/ustreamer/work/.bin:/home/yuri/bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/sbin:/usr/local/bin".  Stop.
===> Compilation failed unexpectedly.
```

mdevaev, 2024-09-04:
It seems like the best thing I can do right now. In the future, I will have to completely revise the build system and maybe use autotools or something else.

closed 2024-09-04 by mdevaev

mryel00, 2025-02-27:
> It's needed for gpio, drm and janus plugin.

I'm wondering, why the dependency check in the `Makefile` does not check for these flags to be set.
If it's only needed for gpio, drm and janus couldn't you check for `WITH_GPIO=1`, `WITH_SYSTEMD=1` and `WITH_JANUS=1`?

I was also a bit confused as this new dependency wasn't added to the readme yet, but that was only an oversight I guess.

mdevaev, 2025-02-27:
I simply forgot about it. I've just added `pkg-config` to README.

As for the optional dependency on `pkg-config`, it's easier to always require it because my Makefile is far from perfect.

mryel00, 2025-02-27:
Ok, thank you for the fast answer and adding it to the `README` 😄

mdevaev, 2025-02-27:
👌