pikvm/ustreamer #287: -latomic isn't needed on FreeBSD

yurivict, 2024-08-16:
We patch it out in the port.

mdevaev, 2024-08-16:
Does it break compilation?

yurivict, 2024-08-16:
Yes:
```
== LD ustreamer-dump.bin
ld: error: unable to find library -latomic
cc: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[2]: *** [Makefile:112: ustreamer.bin] Error 1
gmake[2]: *** Waiting for unfinished jobs....
ld: error: unable to find library -latomic
cc: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[2]: *** [Makefile:117: ustreamer-dump.bin] Error 1
gmake[2]: Leaving directory '/usr/ports/multimedia/ustreamer/work/ustreamer-6.13/src'
gmake[1]: *** [Makefile:46: apps] Error 2
```

mdevaev, 2025-03-05:
Fixed.