pikvm/ustreamer #288: So its hardware encode only for H264?

StuartIanNaylor, 2024-08-28:
I have been at this for some time and thinking its likely me but compiled both.
The config dir has confused me as doesn't seem to exist as its seems set to the etc folder.
Janus seems to get there but running ustreamer `./ustreamer --device=/dev/video0 --h264-sink=demo::ustreamer::h264 --h264-sink-mode=770 --h264-sink-rm` on a Pi5
https://forums.raspberrypi.com/viewtopic.php?p=2139082#p2139126 as didn't know it didn't do hardware h264 encode
Pi 4 does...

Ustreamer
```
-- INFO  [10430.397          ] -- Starting PiKVM uStreamer 6.13 ...
-- INFO  [10430.397          ] -- Using H264-sink: demo::ustreamer::h264
-- INFO  [10430.397          ] -- HTTP: Listening HTTP on [127.0.0.1]:8080
-- INFO  [10430.397          ] -- H264: Initializing encoder ...
-- INFO  [10430.397          ] -- HTTP: Starting eventloop ...
-- INFO  [10430.514          ] -- Using V4L2 device: /dev/video0
-- INFO  [10430.514          ] -- CAP: Using capture type: single-planar
-- INFO  [10430.514          ] -- CAP: Using input channel: 0
-- ERROR [10430.717          ] -- CAP: Requested resolution=640x480 is unavailable
-- INFO  [10430.801          ] -- CAP: Using resolution: 640x360
-- INFO  [10430.801          ] -- CAP: Using format: YUYV
-- INFO  [10430.881          ] -- CAP: Using HW FPS: 0 -> 30 (coerced)
-- INFO  [10430.881          ] -- CAP: Using IO method: MMAP
-- INFO  [10430.881          ] -- CAP: Requested 5 device buffers, got 5
-- INFO  [10431.091          ] -- CAP: Capturing started
-- INFO  [10431.091          ] -- Using JPEG quality: 80%
-- INFO  [10431.091          ] -- Creating pool JPEG with 4 workers ...
-- INFO  [10431.091          ] -- Capturing ...
-- INFO  [10431.155          ] -- H264: Configuring encoder: DMA=1 ...
-- ERROR [10431.155          ] -- H264: Can't open encoder device: No such file or directory
-- ERROR [10431.155          ] -- H264: Encoder destroyed due an error (prepare)

```
Janus
```
Loading plugin 'libjanus_ustreamer.so'...
== ustreamer/main      -- Initializing PiKVM uStreamer plugin 6.13 ...
== ustreamer/config    -- Reading config file '/opt/janus/etc/janus/janus.plugin.ustreamer.jcfg' ...
== ustreamer/audio     -- Probing PCM capture ...
== ustreamer/audio     -- PCM capture is available
```

mdevaev, 2024-08-28:
Yes, it will not work on Pi 5.

StuartIanNaylor, 2024-08-30:
I got one of the RK boards and it has an encoder but its a mess as they have hacked versions of ffmpeg to use it with the BSP.
Mainline in the current Linux-next RK3588 VEPU121/VPU121 support has been submitted.
So with Linux 6.12 I have an alternative to the Pi 5.

The rk3566 uses a cut down version that is 1080p than 4k and even though it shares much its mainline is nowhere near.

Always something but just added for info as did try one.

mdevaev, 2024-08-30:
I don't have any plans to add alternative boards support except Raspberry.