pikvm/ustreamer #293: Ustreamer causing WLAN1 network interface to go DOWN after connection from remote device

zimmertr, 2024-10-25:
I have a Raspberry Pi that I use to run my 3D Printer. This Raspberry Pi runs a piece of software called [Crowsnest](https://github.com/mainsail-crew/crowsnest) which uses ustreamer to stand up a webcam server so that people can monitor their 3D prints while they're occuring. This Raspberry Pi is installed inside my 3D printer's electronics compartment which causes the wireless reception to be poor. To solve this, I've installed a USB Wireless Adapter with an antenna outside of the chassis. 

All of this works great. However, the moment I try and load the webcam feed, ustreamer appears to set the `wlan1` network interface to `DOWN` which breaks the network connectivity to the device. I can see through a monitor connected to the Raspberry Pi that it continues to function. And I can still interact with it just fine. But it loses network connectivity.

At first I thought this was an issue with Crowsnest. But after [filing an issue](https://github.com/mainsail-crew/crowsnest/issues/276) with that project and talking with the dev, it appears to actually be an issue with ustreamer. Here is a video recording that triggers the bug to occur: https://youtu.be/LSgYNcJOXDE

mdevaev, 2024-10-25:
Hello. It's a very weird problem. Ustreamer did nothing with network interfaces so I'm inclined to think that this is a problem with the kernel. There is simply nothing to fix on the part of ustreamer, because it uses normal socket operations.

I know this is not the answer you expected, but I really don't know how I can help. Sorry.

zimmertr, 2024-10-25:
I noticed that `dmesg` was continuously dumping the following error message:

> brcmfmac: brcmf_set_channel: set chanspec fail, reason -52

[image]

This is interesting, because `brcmfmac` is the driver used by the Raspberry Pi's internal `wlan0` NIC

[image]

I set up a second NetworkManager connection profile for `wlan0` and enabled it and rebooted.  Then those errors stopped getting logged.

[image]

Afterwards, I went ahead and started up a ustreamer server again and connected to it over 192.168.30.170 via `wlan1` as usual and it did not crash!

However, using `bmon`, I can see that `wlan0` is the one transferring data continuously, NOT `wlan1`. Despite the fact that I am hitting the `ustreamer` server over `wlan1`'s IP Address.

[image]

Like, I'm happy things are working right now. But something is still wrong. `ustreamer` is only seemingly functional when it can use `wlan0`. And even when I connect to the Raspberry Pi over `wlan1`, I still see that `wlan0` is being used to transfer data. This is problematic because the wireless reception of that NIC is poor, and I can only get 7-11FPS over the 720p stream. Which is what originally led to me purchasing the USB wireless adapter to begin with.

[image]

zimmertr, 2024-10-26:
Seems to have been a `mt7921u` driver issue. Solution: [here](https://github.com/mainsail-crew/crowsnest/issues/276#issuecomment-2439076152).

closed 2024-10-26 by zimmertr

mdevaev, 2024-10-26:
Glad to hear it.