pikvm/ustreamer #297: CAP: Can't set input channel; Microsoft Lifecam VX-800

Schuasda, 2024-12-07:
I try to start ustreamer with default settings, which should work from what i can tell, but i get this:
```shell
Starting PiKVM uStreamer 6.18 ...
HTTP: Listening HTTP on [127.0.0.1]:8080
Using V4L2 device: /dev/video0
CAP: Using capture type: single-planar
CAP: Using input channel: 0
CAP: Can't set input channel
HTTP: Starting eventloop ...
Using V4L2 device: /dev/video0
CAP: Using capture type: single-planar
CAP: Using input channel: 0
CAP: Can't set input channel
```

This is my v4l2-ctl output:
```shell
Driver Info:
        Driver name      : uvcvideo
        Card type        : Microsoft LifeCam VX-800: Micro
        Bus info         : usb-0000:00:14.0-3
        Driver version   : 5.15.167
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : uvcvideo
        Model            : Microsoft LifeCam VX-800: Micro
        Serial           :
        Bus info         : usb-0000:00:14.0-3
        Media version    : 5.15.167
        Hardware revision: 0x00000101 (257)
        Driver version   : 5.15.167
Interface Info:
        ID               : 0x03000002
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : Microsoft LifeCam VX-800: Micro
        Function         : V4L2 I/O
        Flags            : default
        Pad 0x01000007   : 0: Sink
          Link 0x02000010: from remote pad 0x100000a of entity 'Processing 2' (Video Pixel Formatter): Data, Enabled, Immutable
Priority: 2
Video input : 0 (Input 1: ok)
Format Video Capture:
        Width/Height      : 640/480
        Pixel Format      : 'YUYV' (YUYV 4:2:2)
        Field             : None
        Bytes per Line    : 1280
        Size Image        : 614400
        Colorspace        : Default
        Transfer Function : Default (maps to Rec. 709)
        YCbCr/HSV Encoding: Default (maps to ITU-R 601)
        Quantization      : Default (maps to Limited Range)
        Flags             :
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 640, Height 480
        Default     : Left 0, Top 0, Width 640, Height 480
        Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 640, Height 480, Flags:
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 480, Flags:
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Read buffers     : 0

User Controls

                     brightness 0x00980900 (int)    : min=-10 max=10 step=1 default=2 value=2
                       contrast 0x00980901 (int)    : min=0 max=20 step=1 default=10 value=10
                     saturation 0x00980902 (int)    : min=0 max=10 step=1 default=4 value=4
                            hue 0x00980903 (int)    : min=-5 max=5 step=1 default=0 value=0
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=200 step=1 default=130 value=130
                           gain 0x00980913 (int)    : min=32 max=48 step=1 default=34 value=34
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=0 (Disabled)
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=6500 value=6500 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=10 step=1 default=4 value=4
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=1 value=1

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
                                1: Manual Mode
                                3: Aperture Priority Mode
         exposure_time_absolute 0x009a0902 (int)    : min=8 max=16384 step=1 default=512 value=512 flags=inactive
``` 

Am i missing something?
Does this old shitty webcam simply not work with ustreamer? I know it worked on this Linux machine at some point, though i can't remember what i did back then.

mdevaev, 2024-12-10:
Hello. Sometimes it occurs on rare cameras, but I do not know what it is related to. Please try mjpg-streamer, will it work?

Schuasda, 2024-12-23:
Hi,
I am using ustreamer for my Octoprint instance and set it up through an automatic install script for Octoprint.
...which starts a process of ustreamer on its own, which is probably why the camera was simply "blocked" by that existing process.

So long story short, it was my own incompetense which was the issue, the stream works just fine.

closed 2024-12-23 by Schuasda

mdevaev, 2024-12-23:
No problem)