pikvm/ustreamer #298: Audio support from USB webcamera

m-anish, 2024-12-17:
Related to #122 ... I am using ustreamer as part of cncjs, and am able to get a video stream from a webcamera. I was wondering if there's an easy way to also incorporate the audio from the camera's mic too? Running on a rpi 3b+

Thanks.

djkraven, 2024-12-17:
You use VLC. Here is an example of streaming to YouTube.
https://github.com/pikvm/ustreamer/blob/master/docs/youtube.md

mdevaev, 2024-12-18:
Hello. Are you're using janus for streaming?

m-anish, 2024-12-18:
> Hello. Are you're using janus for streaming?

No. I think cncjs just uses ustreamer stright-up https://github.com/cncjs/cncjs-pi-raspbian/blob/master/cncjs_install.sh#L1054

m-anish, 2024-12-18:
Closing this. Reasons:

1. It seems that ustreamer doesn't support h264 input streams 
2. I found another solution for my use case, which is to broadcast over rtsp, and view it in vlc media player. This way, I make an ffmpeg command to send to rtsp server (mediamtx) and just view the stream remotely over vlc.

closed 2024-12-18 by m-anish