pikvm/ustreamer #300: can't install ustreamer on my raspbry pi zero.

eitaar, 2025-01-22:
hello,
I am trying to install ustreamer to my pi and whenever I try to install the libraires, it gives me this error:

`Err:1 http://archive.raspberrypi.com/debian bookworm/main arm64 linux-image-6.6.62+rpt-rpi-v8 arm64 1:6.6.62-1+rpt1
  403  Forbidden [IP: 93.93.135.117 80]
E: Failed to fetch http://archive.raspberrypi.com/debian/pool/main/l/linux/linux-image-6.6.62%2brpt-rpi-v8_6.6.62-1%2brpt1_arm64.deb  403  Forbidden [IP: 93.93.135.117 80]`

can somebody help me with this problem?

mdevaev, 2025-01-23:
Hello. I don't think this problem is related directly to ustreamer. You have some problems with reaching to Raspberry repo. I would suggest you to ask about this problem on their forum: https://forums.raspberrypi.com/

From our side, we can't nothing to do here.

closed 2025-01-24 by mdevaev

eitaar, 2025-01-25:
I was due to the network restrictions. (idk why I couldn't install these specific libraries)
thanks for your help!

mdevaev, 2025-01-25:
You're welcome.

eitaar, 2025-01-25:
is it possible for you to help me again?
I simply did `sudo ./ustreamer` and I got this error.

-- INFO  [401.005          ] -- Starting PiKVM uStreamer 6.27 ...
-- INFO  [401.071          ] -- HTTP: Listening HTTP on [127.0.0.1]:8080`
-- INFO  [401.074          ] -- Using V4L2 device: /dev/video0
-- INFO  [401.076          ] -- CAP: Using capture type: single-planar
-- INFO  [401.077          ] -- CAP: Using input channel: 0
-- INFO  [401.079          ] -- CAP: Using resolution: 640x480
-- INFO  [401.079          ] -- HTTP: Starting eventloop ...
-- INFO  [401.080          ] -- CAP: Using format: YUYV
-- INFO  [401.080          ] -- CAP: Querying HW FPS changing is not supported
-- INFO  [401.081          ] -- CAP: Using IO method: MMAP
-- INFO  [401.085          ] -- CAP: Requested 2 device buffers, got 3
-- ERROR [401.090          ] -- CAP: Can't start capturing: Invalid argument
-- INFO  [401.091          ] -- CAP: Capturing stopped

mdevaev, 2025-01-25:
Which camera do you use?

eitaar, 2025-01-25:
I don't know the specific model because I bought it off AliExpress: [camera](aliexpress.com/item/1005006843276038.html?spm=a2g0o.order_list.order_list_main.5.2444585aIgdqIq&gatewayAdapt=glo2jpn)

information of camera looks like this
````
pi@pizero:~ $ v4l2-ctl -D
Driver Info:
        Driver name      : unicam
        Card type        : unicam
        Bus info         : platform:20801000.csi
        Driver version   : 6.6.62
        Capabilities     : 0xa5a00001
                Video Capture
                Metadata Capture
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x25200001
                Video Capture
                Read/Write
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : unicam
        Model            : unicam
        Serial           :
        Bus info         : platform:20801000.csi
        Media version    : 6.6.62
        Hardware revision: 0x00000000 (0)
        Driver version   : 6.6.62
Interface Info:
        ID               : 0x03000005
        Type             : V4L Video
Entity Info:
        ID               : 0x00000003 (3)
        Name             : unicam-image
        Function         : V4L2 I/O
        Flags            : default
        Pad 0x01000004   : 0: Sink
          Link 0x02000007: from remote pad 0x1000002 of entity 'ov5647 10-0036' (Camera Sensor): Data, Enabled, Immutable
pi@pizero:~ $ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YUYV' (YUYV 4:2:2)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [1]: 'UYVY' (UYVY 4:2:2)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [2]: 'YVYU' (YVYU 4:2:2)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [3]: 'VYUY' (VYUY 4:2:2)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [4]: 'RGBP' (16-bit RGB 5-6-5)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [5]: 'RGBR' (16-bit RGB 5-6-5 BE)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [6]: 'RGBO' (16-bit A/XRGB 1-5-5-5)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [7]: 'RGBQ' (16-bit A/XRGB 1-5-5-5 BE)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [8]: 'RGB3' (24-bit RGB 8-8-8)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [9]: 'BGR3' (24-bit BGR 8-8-8)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [10]: 'RGB4' (32-bit A/XRGB 8-8-8-8)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [11]: 'BA81' (8-bit Bayer BGBG/GRGR)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [12]: 'GBRG' (8-bit Bayer GBGB/RGRG)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
[... 60 lines cut ...]
        [43]: 'GREY' (8-bit Greyscale)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [44]: 'Y10P' (10-bit Greyscale (MIPI Packed))
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [45]: 'Y10 ' (10-bit Greyscale)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [46]: 'Y12P' (12-bit Greyscale (MIPI Packed))
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [47]: 'Y12 ' (12-bit Greyscale)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [48]: 'Y14P' (14-bit Greyscale (MIPI Packed))
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [49]: 'Y14 ' (14-bit Greyscale)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [50]: 'Y16 ' (16-bit Greyscale)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
pi@pizero:~ $ libcamera-hello --list-camera
Available cameras
-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [46.34 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [32.81 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]
````

mdevaev, 2025-01-26:
Ustreamer doesn't work with Pi cameras out of box. Please try this guide: https://github.com/pikvm/ustreamer?tab=readme-ov-file#raspberry-pi-camera-example

eitaar, 2025-01-26:
Hello, I'm sorry to keep asking question to you but could you help me with this as well?
I followed the guide That you given it to me and it worked, but the framerate is extremely low and it shows green screen as well.

The log looks like this:
````
pi@pizero:~/ustreamer $ sudo libcamerify ./ustreamer --encoder=m2m-video --host=0.0.0.0 --port=8080
-- INFO  [274.333          ] -- Starting PiKVM uStreamer 6.27 ...
-- INFO  [274.399          ] -- HTTP: Listening HTTP on [0.0.0.0]:8080
-- INFO  [274.403          ] -- HTTP: Starting eventloop ...
[0:04:34.407906796] [1948] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
[0:04:34.411489215] [1948] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
[0:04:34.414386554] [1948]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+99-1230f78d
[0:04:34.645902634] [1950]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:04:34.666274017] [1950]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
-- INFO  [274.670          ] -- Using V4L2 device: /dev/video0
-- INFO  [274.672          ] -- CAP: Using capture type: single-planar
-- INFO  [274.673          ] -- CAP: Using input channel: 0
[0:04:34.677331311] [1948]  INFO Camera camera.cpp:1197 configuring streams: (0) 640x480-YUYV
[0:04:34.681960852] [1950]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
-- INFO  [274.689          ] -- CAP: Using resolution: 640x480
-- INFO  [274.691          ] -- CAP: Using format: YUYV
-- INFO  [274.692          ] -- CAP: Querying HW FPS changing is not supported
-- INFO  [274.692          ] -- CAP: Using IO method: MMAP
[0:04:34.693341183] [1948]  INFO Camera camera.cpp:1197 configuring streams: (0) 640x480-YUYV
[0:04:34.695880480] [1950]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
-- INFO  [274.706          ] -- CAP: Requested 2 device buffers, got 2
-- INFO  [274.827          ] -- CAP: Capturing started
-- INFO  [274.828          ] -- JPEG-0: Initializing encoder ...
-- INFO  [274.828          ] -- Using JPEG quality: 80%
-- INFO  [274.828          ] -- Creating pool JPEG with 1 workers ...
-- INFO  [274.835          ] -- Capturing ...
-- ERROR [275.348          ] -- CAP: Can't grab HW buffer: Resource temporarily unavailable
-- INFO  [275.439          ] -- Destroying workers pool JPEG ...
-- INFO  [275.462          ] -- CAP: Capturing stopped
================================================================================
-- INFO  [275.469          ] -- Using V4L2 device: /dev/video0
-- INFO  [275.471          ] -- CAP: Using capture type: single-planar
-- INFO  [275.473          ] -- CAP: Using input channel: 0
[0:04:35.479141980] [1948]  INFO Camera camera.cpp:1197 configuring streams: (0) 640x480-YUYV
[0:04:35.485240687] [1950]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
-- INFO  [275.492          ] -- CAP: Using resolution: 640x480
-- INFO  [275.493          ] -- CAP: Using format: YUYV
-- INFO  [275.494          ] -- CAP: Querying HW FPS changing is not supported
-- INFO  [275.495          ] -- CAP: Using IO method: MMAP
[0:04:35.498693246] [1948]  INFO Camera camera.cpp:1197 configuring streams: (0) 640x480-YUYV
[0:04:35.503863845] [1950]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
-- INFO  [275.516          ] -- CAP: Requested 2 device buffers, got 2
-- INFO  [275.638          ] -- CAP: Capturing started
-- INFO  [275.638          ] -- Using JPEG quality: 80%
-- INFO  [275.638          ] -- Creating pool JPEG with 1 workers ...
-- INFO  [275.639          ] -- Capturing ...
[0:04:35.776014386] [1948] ERROR V4L2Compat v4l2_camera_proxy.cpp:682 Failed to clear eventfd POLLIN
[0:04:35.778234643] [1948] ERROR V4L2Compat v4l2_camera_proxy.cpp:682 Failed to clear eventfd POLLIN
[0:04:35.779530793] [1948] ERROR Request request.cpp:480 FrameBuffer already set for stream
[0:04:35.780222873] [1948] ERROR V4L2Compat v4l2_camera.cpp:263 Can't set buffer for request
-- ERROR [275.781          ] -- CAP: Can't release HW buffer=1 (skipped frame): Cannot allocate memory
-- INFO  [275.841          ] -- Destroying workers pool JPEG ...
-- INFO  [275.862          ] -- CAP: Capturing stopped
================================================================================
-- INFO  [275.865          ] -- Using V4L2 device: /dev/video0
-- INFO  [275.867          ] -- CAP: Using capture type: single-planar
-- INFO  [275.868          ] -- CAP: Using input channel: 0
[0:04:35.873908731] [1948]  INFO Camera camera.cpp:1197 configuring streams: (0) 640x480-YUYV
[0:04:35.877135105] [1950]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
-- INFO  [275.883          ] -- CAP: Using resolution: 640x480
-- INFO  [275.883          ] -- CAP: Using format: YUYV
-- INFO  [275.883          ] -- CAP: Querying HW FPS changing is not supported
-- INFO  [275.883          ] -- CAP: Using IO method: MMAP
[... cut ...]
[0:04:36.148321533] [1966] ERROR Request request.cpp:480 FrameBuffer already set for stream
[0:04:36.148858595] [1966] ERROR V4L2Compat v4l2_camera.cpp:263 Can't set buffer for request
-- ERROR [276.150          ] -- CAP: Can't release HW buffer=1: Cannot allocate memory
-- INFO  [276.248          ] -- Destroying workers pool JPEG ...
^C-- INFO  [276.273          ] -- CAP: Capturing stopped
-- INFO  [276.275          ] -- ===== Stopping by SIGINT =====
-- INFO  [276.277          ] -- JPEG-0: Destroying encoder ...
-- INFO  [276.277          ] -- Bye-bye
````