pikvm/ustreamer #302: An error after some updates.

themoonwatcher, 2025-02-03:
Hello and thanks for making the fastest streamer that I know so far!

I'm using it on RPI Zero 2w. It was working fine (excluding the buffer error), but after some recent updates it stopped:

```
$ sudo libcamerify -d /home/s/ustreamer/./ustreamer --encoder=m2m-video --host=0.0.0.0 --port=80 -r 1920x1080 -b 1 -l
-- INFO  [367.159          ] -- Starting PiKVM uStreamer 6.29 ...
-- INFO  [367.228          ] -- HTTP: Listening HTTP on [0.0.0.0]:80
-- INFO  [367.228          ] -- HTTP: Starting eventloop ...
[0:06:07.230036969] [1153] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
[0:06:07.230159156] [1153] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
[0:06:07.230257437] [1153]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+50-83cb8101
[0:06:07.279237139] [1155]  WARN CameraSensorProperties camera_sensor_properties.cpp:473 No static properties available for 'imx708_wide'
[0:06:07.279328597] [1155]  WARN CameraSensorProperties camera_sensor_properties.cpp:475 Please consider updating the camera sensor properties database
[0:06:07.388578620] [1155]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:06:07.392516569] [1155]  WARN CameraSensor camera_sensor_legacy.cpp:501 'imx708_wide': No sensor delays found in static properties. Assuming unverified defaults.
[0:06:07.394356352] [1155]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:06:07.394829891] [1153] DEBUG V4L2Compat v4l2_compat_manager.cpp:79 Started camera manager
[0:06:07.395009942] [1153] DEBUG V4L2Compat v4l2_camera_proxy.cpp:52 [] open()
ustreamer: ../include/libcamera/controls.h:189: T libcamera::ControlValue::get() const [with T = long int; typename std::enable_if<((! libcamera::details::is_span<U>::value) && (! std::is_same<std::__cxx11::basic_string<char>, typename std::remove_cv< <template-parameter-1-1> >::type>::value)), std::nullptr_t>::type <anonymous> = nullptr]: Assertion `type_ == details::control_type<std::remove_cv_t<T>>::value' failed.
Aborted
```

Any idea why?

```
$ uname -a
Linux rfish 6.1.77-v8+ #1768 SMP PREEMPT Tue Jun  4 14:47:07 BST 2024 aarch64 GNU/Linux

```

mdevaev, 2025-02-03:
Hello! As I see, the problem is in libcamerify. Try to use the fresh version of it from sources. I didn't change ustreamer's capture chain a long time, so it can't be related with our updates.

mdevaev, 2025-02-03:
Okay! Well, since the problem isn't with ustreamer, I'm closing this issue. I hope the will fix libcamera soon.

closed 2025-02-03 by mdevaev