pikvm/ustreamer #305: The janus plugin fails to compile because of a missing CFLAG

yurivict, 2025-02-28:
The error is:
```
The ustream project that uses janus fails to find the janus headers:
In file included from src/acap.c:43:
In file included from src/logging.h:23:
/usr/local/include/janus/plugins/plugin.h:162:10: fatal error: 'refcount.h' file not found
162 | #include "refcount.h"
|              ^~~~~~~~~~~~
1 error generated.

```

This is because ```-I$(PREFIX)/include/janus``` is missing.

mdevaev, 2025-02-28:
Technically, I think this is a problem on the janus side.  I use this patch when building Janus:

```
sed -i -e 's|^#include "refcount.h"$|#include "../refcount.h"|g' "$pkgdir/usr/include/janus/plugins/plugin.h"
```

yurivict, 2025-02-28:
I see. I used the janus package.
This should probably be reported as a bug to the Janus project.

I also asked them to add the .pc file: https://github.com/meetecho/janus-gateway/issues/3523

Please close it then if you feel that the problem is really in Janus.

mdevaev, 2025-02-28:
Thanks. Let's wait of Janus answer and will see.

yurivict, 2025-03-05:
Thanks for letting me know.
I'll remove the workaround in the port once this change will make it into the janus package.

mdevaev, 2025-03-05:
👌

closed 2025-03-05 by mdevaev