pikvm/ustreamer #306: sys/sysmacros.h doesn't seem to be needed in src/libs/drm/drm.c

yurivict, 2025-02-28:
It builds without this line on FreeBSD.
Is this line really needed?

mdevaev, 2025-02-28:
Yes, it's for `minor()` macro. Btw I don't think you need to build uStreamer with `WITH_V4P=1` option at all, it was made only for PiKVM V4 Plus and it's guaranteed to work only there, it's not needed or supported for anything else.

However, for formality, I could add ifndef. What do you say?

yurivict, 2025-02-28:
> However, for formality, I could add ifndef. What do you say?

I agree.

yurivict, 2025-02-28:
> Btw I don't think you need to build uStreamer with WITH_V4P=1 option at all, it was made only for PiKVM V4 Plus and it's guaranteed to work only there, it's not needed or supported for anything else.

The README doesn't seem to have descriptions of optional dependencies.
Without having a description of what this option does and what it is intended for, I just added the corresponding option to the FreeBSD port.

mdevaev, 2025-02-28:
Okay, fixed.

mdevaev, 2025-02-28:
PS: I did not document this option precisely for reasons that it is for internal use.

closed 2025-02-28 by mdevaev