pikvm/ustreamer #307: Select() timeout with PiKVM and C790

cptcharisma, 2025-03-16:
`[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [40.881          ] -- Using V4L2 device: /dev/kvmd-video
[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [40.882          ] -- CAP: Using capture type: single-planar
[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [40.882          ] -- CAP: Using input channel: 0
[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [40.901          ] -- CAP: Detected DV-timings: 1920x1080p50.00, pixclk=115024000, vsync=26, hsync=160
[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [40.951          ] -- CAP: Using resolution: 1920x1080
[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [40.951          ] -- CAP: Using format: UYVY
[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [40.951          ] -- CAP: Querying HW FPS changing is not supported
[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [40.951          ] -- CAP: Using IO method: MMAP
[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [40.977          ] -- CAP: Requested 6 device buffers, got 6
[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [40.990          ] -- CAP: Capturing started
[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [40.990          ] -- Using JPEG quality: 80%
[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [40.990          ] -- Creating pool JPEG with 3 workers ...
[2025-01-14 20:14:51 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [40.991          ] -- Capturing ...
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- ERROR [41.992          ] -- CAP: Device select() timeout
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.092          ] -- Destroying workers pool JPEG ...
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.129          ] -- CAP: Capturing stopped
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.146          ] -- Using V4L2 device: /dev/kvmd-video
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.146          ] -- CAP: Using capture type: single-planar
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.146          ] -- CAP: Using input channel: 0
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.156          ] -- CAP: Detected DV-timings: 1920x1080p50.00, pixclk=115024000, vsync=26, hsync=160
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.199          ] -- CAP: Using resolution: 1920x1080
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.199          ] -- CAP: Using format: UYVY
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.199          ] -- CAP: Querying HW FPS changing is not supported
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.199          ] -- CAP: Using IO method: MMAP
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.226          ] -- CAP: Requested 6 device buffers, got 6
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.238          ] -- CAP: Capturing started
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.238          ] -- Using JPEG quality: 80%
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.238          ] -- Creating pool JPEG with 3 workers ...
[2025-01-14 20:14:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [42.239          ] -- Capturing ...`

I get no video signal on the web portal. The keyboard and mouse control work.
`# dmesg | egrep 'tc35|1-1.4|uvc'
egrep: warning: egrep is obsolescent; using grep -E
[    0.041027] platform fe801000.csi: Fixed dependency cycle(s) with /soc/i2c0mux/i2c@1/tc358743@f
[    6.938192] platform fe801000.csi: Fixed dependency cycle(s) with /soc/i2c0mux/i2c@1/tc358743@f
[    7.122366] tc358743 10-000f: tc358743 found @ 0x1e (i2c-22-mux (chan_id 1))
# systemctl status kvmd-tc358743
* kvmd-tc358743.service - PiKVM - EDID loader for TC358743
     Loaded: loaded (/usr/lib/systemd/system/kvmd-tc358743.service; enabled; preset: disabled)
     Active: active (exited) since Tue 2025-01-14 20:14:19 UTC; 1 month 30 days ago
 Invocation: db4e4b29e5624891b67e46f4672441d1
    Process: 468 ExecStart=/usr/bin/v4l2-ctl --device=/dev/kvmd-video --set-edid=file=/etc/kvmd/tc358743-edid.hex --info-edid (code=exited, status=0/SUCCESS)
   Main PID: 468 (code=exited, status=0/SUCCESS)
        CPU: 28ms

Jan 14 20:14:19 pikvm systemd[1]: Starting PiKVM - EDID loader for TC358743...
Jan 14 20:14:19 pikvm v4l2-ctl[468]: CTA-861 Header
Jan 14 20:14:19 pikvm v4l2-ctl[468]:   IT Formats Underscanned: no
Jan 14 20:14:19 pikvm v4l2-ctl[468]:   Audio:                   no
Jan 14 20:14:19 pikvm v4l2-ctl[468]:   YCbCr 4:4:4:             no
Jan 14 20:14:19 pikvm v4l2-ctl[468]:   YCbCr 4:2:2:             no
Jan 14 20:14:19 pikvm systemd[1]: Finished PiKVM - EDID loader for TC358743.

# v4l2-ctl --query-dv-timings
        Active width: 1920
        Active height: 1080
        Total width: 2080
        Total height: 1106
        Frame format: progressive
        Polarities: -vsync -hsync
        Pixelclock: 115024000 Hz (50.00 frames per second)
        Horizontal frontporch: 0
        Horizontal sync: 160
        Horizontal backporch: 0
        Vertical frontporch: 0
        Vertical sync: 26
        Vertical backporch: 0
        Standards:
        Flags:
`

I tried the discord but to no avail. Looking for some guidance.

mdevaev, 2025-03-16:
You have "no signal" or the black screen?

cptcharisma, 2025-03-16:
No signal

mdevaev, 2025-03-16:
Okay, I need more info. Which PiKVM device do you use?

cptcharisma, 2025-03-16:
I use raspberry pi 4 model b. I have C790 connected to the csi port. I have a usb splitter board for delivering power to the pi and passing hid to the target computer. The target is a windows machine that detects a PiKVM monitor and is running at 1080p@50hz. Let me know what more details I can provide and thank you for looking into this.

mdevaev, 2025-03-17:
Okay, this PiKVM-specific support question, so please use the discord. Our support guys will help you with it.

https://discord.gg/bpmXfz5

cptcharisma, 2025-03-17:
I will try that. Thanks!

closed 2025-03-17 by cptcharisma