pikvm/ustreamer #308: Unable to Read Frames from uStreamer Unix Socket in Golang

iamsavani, 2025-03-17:
I'm trying to integrate **uStreamer** with a **Go (Golang)** application to capture raw frames via a **Unix socket** and send them over **WebRTC** using Pion. However, I am unable to read frames from the socket, and the connection either **closes immediately** or **returns EO**F.

**Start uStreamer with the following command:**
`ustreamer --device=/dev/video1 --encoder=hw --unix=/run/ustreamer.sock --unix-rm --format=yuyv --port=8002 --host=0.0.0.0 --unix-mode=0660`

**Run my Go code to read frames from the socket:**
`
conn, err := net.Dial("unix", "/run/ustreamer.sock")
    if err != nil {
        log.Fatalf("Failed to connect to uStreamer socket: %v", err)
    }
    defer conn.Close()

    inboundPacket := make([]byte, frameSize)

    for {
        n, err := conn.Read(inboundPacket)
        if err != nil {
            log.Printf("Error during read: %v", err)
            return
        }
        // send frame to the channel
    }
`

While running above code. I am getting this error `Error during read`

**Log**
`
-- INFO  [42422.581    stream] -- Device fd=8 opened
-- INFO  [42422.581    stream] -- Using input channel: 0
-- ERROR [42422.581    stream] -- Can't set input channel
-- INFO  [42422.582    stream] -- Device fd=8 closed
-- INFO  [42422.582    stream] -- Sleeping 1 seconds before new stream init ...
`

I am not sure if I am using uStreamer correctly, but the idea is to use it with a different WebRTC library other than Janus.

mdevaev, 2025-03-17:
* uStreamer does not provide raw interface over unix socket, it's http-over-unix-domain-socket.
* "Can't set input channel" is a configuration problem.

iamsavani, 2025-03-17:
Thank you for your response! I appreciate the clarification about uStreamer’s Unix socket being HTTP-over-UDS.

My goal is to integrate uStreamer natively without relying on Janus, allowing for more flexibility in using different programming languages like Go and Node.js. I’d love to explore how to efficiently access and process the video stream directly within these environments.

If there are any recommended approaches or best practices for achieving this, I’d greatly appreciate your insights!

mdevaev, 2025-03-17:
You can use shared memory to access to frames. But I don't have Go experience so I can't tell will ir wrong fine or not.

iamsavani, 2025-03-17:
I’m working on a C library that reads frames from a shared memory segment (/dev/shm/test.h264). I intend to use this with Go via cgo bindings. Below is what I have so far, but I’m unsure if I’m handling shared memory access and frame reading correctly.

Key concerns:

- Am I correctly mapping and reading frames from shared memory?
- Is there a better way to handle synchronization or detect frame boundaries?
- How can I ensure that I always get a complete frame and not partial/corrupt data?

Here’s the C code I’ve written:

```
#include <stdio.h>
#include <stdlib.h>
#include <fcntl.h>
#include <sys/mman.h>
#include <sys/stat.h>
#include <unistd.h>
#include <stdint.h>
#include <string.h>

#define SHM_PATH "/dev/shm/test.h264"

static uint8_t *shm_ptr = NULL;
static size_t shm_size = 0;

// Initialize shared memory mapping
int shm_init() {
    int fd = open(SHM_PATH, O_RDONLY);
    if (fd < 0) {
        perror("Failed to open shared memory");
        return -1;
    }

    struct stat sb;
    if (fstat(fd, &sb) == -1) {
        perror("Failed to stat shared memory");
        close(fd);
        return -1;
    }
    shm_size = sb.st_size;

    shm_ptr = (uint8_t *)mmap(NULL, shm_size, PROT_READ, MAP_SHARED, fd, 0);
    close(fd);

    if (shm_ptr == MAP_FAILED) {
        perror("Failed to mmap shared memory");
        return -1;
    }

    printf("Shared memory initialized, size: %zu bytes\n", shm_size);
    return 0;
}

// Read a frame from shared memory
int read_frame(uint8_t *buffer, size_t max_size) {
    if (!shm_ptr) {
        fprintf(stderr, "Shared memory not initialized\n");
        return -1;
    }

    size_t read_size = (shm_size < max_size) ? shm_size : max_size;
    memcpy(buffer, shm_ptr, read_size);

    return read_size;
}

// Cleanup shared memory
void shm_cleanup() {
    if (shm_ptr) {
        munmap(shm_ptr, shm_size);
        shm_ptr = NULL;
    }
}
```