pikvm/ustreamer #309: CAP: Device select() timeout - running multiple copies

jaymemaurice, 2025-05-22:
I have 4 5MP cameras hooked up to a Raspberry Pi Zero 2W with the Waveshare USB Hat (B) running Alpine Linux. 

I have built from [2717248](https://github.com/pikvm/ustreamer/commit/271724858176074b5f94dfe2bbb066f1f6db38aa)

I don't expect all 4 cameras to stream simultaneously, but I didn't expect the CAP: Device select() timeout errors before launching any streams.

To reproduce:
ustreamer --format=mjpeg  --device=/dev/video0 --host=192.168.2.144 --port=8000 -r 3264x2448 -f 2 --device-timeout=10&
ustreamer --format=mjpeg  --device=/dev/video4 --host=192.168.2.144 --port=8004 -r 3264x2448 -f 2 --device-timeout=10&
ustreamer --format=mjpeg  --device=/dev/video6 --host=192.168.2.144 --port=8006 -r 3264x2448 -f 2 --device-timeout=10&
ustreamer --format=mjpeg  --device=/dev/video2 --host=192.168.2.144 --port=8002 -r 3264x2448 -f 2 --device-timeout=10&

When I turn on debug I see that it appears that the devices all become active on the USB bus despite nobody watching. I assume there isn't enough CPU power/USB bandwidth to support 4 5MP cameras simultaneously? 

-- VERB  [2037.505          ] -- JPEG: Passed encoding because nobody is watching
-- DEBUG [2037.510          ] -- CAP: Releasing HW buffer=4 ...
-- DEBUG [2037.510          ] -- CAP: HW buffer=4 released
-- DEBUG [2037.558          ] -- CAP: Device select() --> 1; has_read=1, has_error=0
-- DEBUG [2037.558          ] -- CAP: Grabbing hw buffer ...
-- DEBUG [2037.558          ] -- CAP: Grabbed HW buffer=0: bytesused=704748, grab_ts=2037.493, latency=0.065, skipped=0
-- DEBUG [2037.558          ] -- CAP: Calling select() on video device ...

Is there a way we can make the video device come online only when there is an active streamer?

mdevaev, 2025-05-22:
Hello. You're right about USB bandwidth, one bus just can't handle so much data. There is nothing we can do here.

closed 2025-05-24 by mdevaev

jaymemaurice, 2025-05-25:
okay fair enough but I don't actually need full streaming from each camera. 
I need the stream to update at least once every 5 minutes.

I've launched 4 ustreamer instances with the following parameters:
```
 8339 root      0:00 supervise-daemon camera1 --start --respawn-delay 2 --respawn-max 5 --respawn-period 1800 /usr/local/bin/ustreamer -- --format=mjpeg --device=/dev/video0 --host=192.168.2.144 --port=8000 -r 3264x2448 -f 1 -n -l --device-timeout 20 -w 2 -K 0
 8396 root      0:00 supervise-daemon camera2 --start --respawn-delay 2 --respawn-max 5 --respawn-period 1800 /usr/local/bin/ustreamer -- --format=mjpeg --device=/dev/video2 --host=192.168.2.144 --port=8002 -r 3264x2448 -f 1 -n -l --device-timeout 20 -w 2 -K 0
 8453 root      0:00 supervise-daemon camera3 --start --respawn-delay 2 --respawn-max 5 --respawn-period 1800 /usr/local/bin/ustreamer -- --format=mjpeg --device=/dev/video4 --host=192.168.2.144 --port=8004 -r 3264x2448 -f 1 -n -l --device-timeout 20 -w 2 -K 0
 8510 root      0:00 supervise-daemon camera4 --start --respawn-delay 2 --respawn-max 5 --respawn-period 1800 /usr/local/bin/ustreamer -- --format=mjpeg --device=/dev/video6 --host=192.168.2.144 --port=8006 -r 3264x2448 -f 1 -n -l --device-timeout 20 -w 2 -K 0
```

then I have another script resetting the USB ports one at a time...
```
while true; do for i in 1 2 3 4; do echo disabling usb1-1.$i; echo 0 > /sys/bus/usb/devices/1-1.$i/authorized; sleep 3; echo enabling usb 1-1.$i; echo 1 > /sys/bus/usb/devices/1-1.$i/authorized; sleep 60; done; done;
```

This sort of works in that 3 of the 4 cameras are always working....

However I would expect -K 0 should make it such that the kicked/resetting camera shows the last frame. Instead I get the black disconnected screen. Is the -K 0 option only applicable when transcoding or is there a bug here?

mdevaev, 2025-05-25:
`-K`/`--last-as-blank` option is deprecated for a year or so. I had to remove this logic for the sake of optimizing and improving FPS.