pikvm/ustreamer #313: Low FPS with ustreamer, Great Performance mjpg-streamer

Zachary-Howell, 2025-07-12:
I am able to get 30fps when I stream to my browser at 1920x1080 using mjpg-streamer. 
When I attempt to stream using ustreamer, I get ~1 fps at all resolutions unless I use YUYV, which prevents me from using higher resolutions due to bandwidth limitations.

- Raspberry Pi 5 running Mainsail OS
- Logitech C920 HD Pro

Example ustreamer command:
```
admin@mainsailos:~ $ ~/ustreamer/ustreamer \
  --device=/dev/video0 \
  --format=MJPEG \
  --resolution=1920x1080 \
  --desired-fps=30 \
  --log-level=1 \
  --host=0.0.0.0 \
  --port=8080
```

Example mjpg-streamer command:
'''
mjpg_streamer -i "input_uvc.so -r 1920x1080 -f 30" -o "output_http.so -p 8080 -w ./www"
'''

Zachary-Howell, 2025-07-13:
Sure! Here is a level 2 log. Let me know if there's any other info I can provide that would be useful.

[ustreamer_fail.log](https://github.com/user-attachments/files/21206347/ustreamer_fail.log)

mdevaev, 2025-07-15:
Please show `v4l2-ctl --list-formats-ext`

mdevaev, 2025-07-16:
Try to run ustreamer with `--debug`.

Zachary-Howell, 2025-07-18:
Fantastic, that seems to have worked! Thank you!

closed 2025-07-18 by Zachary-Howell

tastyratz, 2025-09-15:
For anyone who comes here to find this reference;
On a C920 in Crowsnest/Mainsail with a Pi5
custom_flags: --format=YUYV 
brought me from 2fps to around 10fps

custom_flags: --allow-truncated-frames --encoder=HW
Brought me to 20fps ( must remove yuyv)