pikvm/ustreamer #314: compilation error ?

ski007, 2025-07-19:
`make
make apps
make[1]: Wejście do katalogu '/root/ustreamer-master'
make -C src
make[2]: Wejście do katalogu '/root/ustreamer-master/src'
-- CC libs/base64.c
-- CC libs/capture.c
-- CC libs/fpsi.c
-- CC libs/frame.c
-- CC libs/frametext.c
-- CC libs/frametext_font.c
-- CC libs/logging.c
-- CC libs/memsink.c
-- CC libs/memsinksh.c
-- CC libs/options.c
-- CC libs/queue.c
-- CC libs/ring.c
-- CC libs/signal.c
-- CC libs/tc358743.c
-- CC libs/unjpeg.c
-- CC ustreamer/blank.c
-- CC ustreamer/data/favicon_ico.c
-- CC ustreamer/data/index_html.c
-- CC ustreamer/encoder.c
-- CC ustreamer/encoders/cpu/encoder.c
-- CC ustreamer/encoders/hw/encoder.c
-- CC ustreamer/http/mime.c
ustreamer/http/mime.c:27:10: fatal error: event2/util.h: Nie ma takiego pliku ani katalogu
   27 | #include <event2/util.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:135: build/ustreamer/http/mime.o] Błąd 1
make[2]: Opuszczenie katalogu '/root/ustreamer-master/src'
make[1]: *** [Makefile:67: apps] Błąd 2
make[1]: Opuszczenie katalogu '/root/ustreamer-master'
make: *** [Makefile:57: all] Błąd 2
`
can this be fixed somehow?

mdevaev, 2025-07-19:
You need to install libevent. Please read the README file with installation guide.

ski007, 2025-07-20:
You're right! ...sorry, everything works perfectly :-)

mdevaev, 2025-07-20:
No problem.